OneOfEleven/uv-k5-firmware-custom #10: DTMF Decode

spm81, 2023-09-18:
DTMF Decode works with QuanShang radio but with baofeng for exemple don't decode all characters.

Keep the good job ;)

OneOfEleven, 2023-09-18:
mmm you're right, never knew that. The QS firmware must be the same too, I'll look at the DTMF settings to the radio chip, although we don't have a detailed data sheet for this radios chip :(

OneOfEleven, 2023-09-19:
There are 16 DTMF coefficient values that the firmware sets in the BK4819 radio chip, knowing what exactly they do and how best to adjust them I'm sure would fix the DTMF decoder.

However, we don't have that info, we need detailed info on the chips registers (we have bits of info but not all) :(

lohtse, 2023-09-19:
could you add this to makefile please 

 ENABLE_DTMF_DECODER l=0 

see in group for the reason why

OneOfEleven, 2023-09-19:
What groups that ?

OneOfEleven, 2023-09-19:
??

I don't use telegram

lohtse, 2023-09-19:
[image]

lohtse, 2023-09-19:
when dtmf is enable it prints out link that for some when using repeaters.. no issue when disable

OneOfEleven, 2023-09-19:
The DTMF decoder is now enabled/disabled in the menu, look for "D-LIVE"

lohtse, 2023-09-19:
ar ok

OneOfEleven, 2023-09-25:
> DTMF Decode works with QuanShang radio but with baofeng for exemple don't decode all characters.
> 
> Keep the good job ;)

I've tuned the DTMF decoder a bit, it now works with the wuoxun radio I have here, maybe you can try the decoder with your other radios ? .. let me know how it goes if you do

spm81, 2023-09-26:
The DTMF decoder is now enabled/disabled in the menu, look for "D-LIVE"

D-LIVE is ON but nothing appeared :(((

OneOfEleven, 2023-09-26:
Have you the ENABLE_AUDIO_BAR set to '1' ?

If so, set it to '0' to disable it, recompile, or disable the MIC bar menu option. The displays center line can only be used one or the other function. Also have to be in FM mode.

spm81, 2023-09-26:
ok , i see... i will try it and send you a feedback later ... thanks..... a lot :)

spm81, 2023-09-26:
i disable MIC bar option and DTMF decoder worked :)))))))))))))
I try with the Baofeng UV-5R again and.................... ALL KEYS WORKING 🥇 
Thanks once again..... Keep the good job 💯 %

OneOfEleven, 2023-09-26:
Good good, was interested to know if it now works OK for anyone else.

closed 2023-09-26 by OneOfEleven