OneOfEleven/uv-k5-firmware-custom #3: compiling  using VSCODE

lohtse, 2023-09-18:
when ever I edit the makefile in VSCode the changes are not applied... if I change the makefile using notepad those changes then work in vscode(as well as win_make.bat)

OneOfEleven, 2023-09-18:
Sounds like VSCode isn't saving the make file before compiling, I don't know without trying myself.

I haven't got VSCode configured to build the project, was it easy to setup I wonder ?

lohtse, 2023-09-18:
> Sounds like VSCode isn't saving the make file before compiling, I don't know without trying myself.
> 
> I haven't got VSCode configured to build the project, was it easy to setup I wonder ?

in general it is very easy  just install as required in the gcc stuff and you should be ready to go.... Compiles other k5 branches fine

wrcrooks, 2023-10-24:
Works fine for me compiling via VSCode. I'm using the extension "Makefile Tools" (v0.7.0) from Microsoft

closed 2023-10-24 by OneOfEleven