OneOfEleven/uv-k5-firmware-custom #4: AM receive registers above airband

pdrbsts, 2023-09-18:
Your new setting improved airband reception by a lot.
Can you maintain the original settings above 225 MHz and use your new settings below?

OneOfEleven, 2023-09-18:
Yes that's easy to do. What happens for you above 225MHz I wonder ?

OneOfEleven, 2023-09-18:
Seems the RX gain drops a fair bit after 226.4MHz, so I've set it to use the original gain setting above that frequency now. See how it goes.

M7OCM, 2023-09-19:
Testing today so will let you know 111. Much appreciated.

M7OCM, 2023-09-19:
Just posted a vid on OEFW Telegram 111, perfect sir thank you so much.

OneOfEleven, 2023-09-19:
I don't have/use Telegram (so can't see your vid :()
I don't use any social media (I tried FB once years ago but quickly left it), though I do like youtube for tech/sci vids.

OneOfEleven, 2023-09-19:
The AM reception has problems, we (I) don't have the detailed info on the BK chip to fix it :(

M7OCM, 2023-09-19:
Copy that 111 Ill pop up a link from YT .

M7OCM, 2023-09-19:
https://youtu.be/bDg3eTHMj5E?si=bip0ptKNyIeOKT9r   yep the chip is full of mysteries lol that said very happy with your mods and look its very pleasing to the eye and actually works bonus!

DE8MSH, 2023-09-19:
Hello OneOfEleven. 

Is the "RSSI related GAIN (LNA etc) implemented in general?

OneOfEleven, 2023-09-19:
I haven't yet passed the gain change settings over to the RSSI reading read from the BK chip, but  it does need doing (to keep the RSSI calibrated). I wanted to see if the gain changes are OK to keep (or not) first.

Presume that's what you mean ?

DE8MSH, 2023-09-20:
I mean if RSSI val is high LNA (etc) gain should be lowered. Vice versa. 'cause: here one station has lower signal strength. it's little bit noisy but understandable. then a very strong station answers. signal is overloaded and makes demodulated voice disunderstandable.

OneOfEleven, 2023-09-20:
Is that in AM mode or FM ?

Increasing the front end gain like I have above 226.4MHz does have the side effect of reducing the RX's dynamic range (it's how things work). I may have to either dynamically adjust the front end gain or bring the gain back down to where it was.

Wonder why they don't publish a full datasheet on the BK chip .. grrr

DE8MSH, 2023-09-20:
yes. many peeoples thinking so. we need the whole bk specs.... i mean for AM mode. if rssi is for eg -90 dB set LNA etc to n and if RSSI is -40 dB set LNA and co to other n
and when to meassure RSSI: if AM transmission begins. i think Radtel 890 (?) does it like this. if -20 dB transmission beginns mudolation sounds awfull / distorted. bus after 1/2 or 1 second it sounds good surprisingly.

DE8MSH, 2023-09-22:
Hi 1O11. I think your working on https://github.com/OneOfEleven/uv-k5-firmware-custom/issues/4#issuecomment-1727538236

Seems to go into the right direction. I had following situation: 1st station relatively low sig. Noisy but understandable. Then strong station comes up after that (normal here on a specific channel). But K5 seems not to get right RSSI value and then modulation starts to fade strangely.

FYI.

OneOfEleven, 2023-09-22:
It's OK, if your trying it just ignore the RSSI reading for now, I'll take care of that once I've got it working properly.

I've deleted TG, I don't like social media, I deleted the account and cleaned it all out. I'll get back to doing my own thing with the firmware shortly, was more enthusiastic about helping with the project before I tried TG, but social media for me puts a big dampener on life.

DE8MSH, 2023-09-22:
TG: It's fully alright 👍

RSSI: To me it's a good solution! The only problem is the fast up and down fading. The fadgi solution is not perfect at all, too. Why: Modulation is 20dB more quiet! I have to push volume knob fully to the right. You solution is much more louder. You can have a look at quickfading maybe? If you like? I'll test. :)

DE8MSH, 2023-09-22:
I have both here. Very strong sigs and less strong sigs. So I can test it. :)

Raul-2017, 2023-09-22:
Congrats for your fresh ideas improving AM reception.   Waiting for your releases to be checked in our  UVK5s and enjoy...  :-)

OneOfEleven, 2023-09-22:
Thanks, it still needs more work.

I'm guessing it's kinda working now ?  I'm lacking regular strong AM signals to test here, so it's taking me a while to optimize and make stable (no gain hunting).

DE8MSH, 2023-09-23:
Hi 1O11. thank you for not giving up dynamic RSSI/gain way :)

im just testing https://github.com/OneOfEleven/uv-k5-firmware-custom/commit/76aa4b893a156056990777f9983318277b319a10 and noting some values. some are are too strong and am is demodulating badly. 

2 be continued...

closed 2023-09-24 by OneOfEleven