OneOfEleven/uv-k5-firmware-custom #8: commit

DE8MSH, 2023-09-18:
Hi.

This comes up at commit 06722a2:

`ui/main.c: In function 'UI_DisplayMain':
ui/main.c:465:7: error: 'center_line_is_free' undeclared (first use in this function)
  465 |   if (center_line_is_free)
      |       ^~~~~~~~~~~~~~~~~~~
ui/main.c:465:7: note: each undeclared identifier is reported only once for each function it appears in
make: *** [Makefile:240: ui/main.o] Fehler 1
`

Makefile is

`ENABLE_SWD                    := 0

ENABLE_OVERLAY                := 1
ENABLE_UART                   := 1
ENABLE_AIRCOPY                := 0
ENABLE_FMRADIO                := 0
ENABLE_NOAA                   := 0
ENABLE_VOICE                  := 0
ENABLE_ALARM                  := 0
ENABLE_BIG_FREQ               := 0
ENABLE_KEEP_MEM_NAME          := 1
ENABLE_CHAN_NAME_FREQ         := 1
ENABLE_WIDE_RX                := 1
ENABLE_TX_WHEN_AM             := 0
ENABLE_CTCSS_TAIL_PHASE_SHIFT := 0
ENABLE_MAIN_KEY_HOLD          := 1
ENABLE_BOOT_BEEPS             := 0
ENABLE_DTMF_DECODER           := 0
ENABLE_COMPANDER              := 0
#ENABLE_SINGLE_VFO_CHAN       := 1
#ENABLE_BAND_SCOPE            := 1`

DE8MSH, 2023-09-18:
Thank you very much vor having a look at it 👍👍👍👍👍

OneOfEleven, 2023-09-18:
I think I've now fixed that for you. Let me know if any problem with it.

DE8MSH, 2023-09-18:
Thank you again!!! :)

closed 2023-09-18 by DE8MSH