OneOfEleven/uv-k5-firmware-custom #9: [Proposal] Extend on-screen DTMF decoder with a T9 parsing capability

LZ2DMV, 2023-09-18:
Since there is already a T9 library written for another K5 firmware - https://github.com/piotr022/UV_K5_playground/blob/main/libs/keyboard/t9.hpp, perhaps you could extend the current on-screen DTMF decoder with a T9 text parsing capability.

For example, if a DTMF sequence looks like a parsable T9 text, show the parsed text instead of the DTMF digits. Ideally, this should be toggable with a separate switch from the one used for the DTMF decoder.

OneOfEleven, 2023-09-18:
Sounds easy to do, though I can't quite work out what T9 is ?

LZ2DMV, 2023-09-18:
https://en.wikipedia.org/wiki/T9_(predictive_text)

Basically, typing text with digits from 0 to 9 - traditionally, on a phone, and in our case - from a radio. :)

OneOfEleven, 2023-09-18:
Ah, you mean each of the number digits having 3 or letters to choose from that you can access by multiple number presses ?

OneOfEleven, 2023-09-18:
I did think about that before, but though as the buttons have no letters printed on them it would be difficult (for those of my age anything) to remember the exact letter layout

LZ2DMV, 2023-09-18:
Yes, it is a shame that Quansheng didn't printed the letters like the Baofeng does.

One way to work around that will be to mod the DTMF sending input dialog that opens with the star (*) key to actually behave just like in ths Messenger app i linked above - upon a keypress, show the first letter for that key, and if the keypress repeats, exchange it for the next letter in that key, and so on.

You can actually borrow the logic from the Messenger app.

The display should show the letter when in DTMF input mode (* key), but when PTT is pressed, DTMF sequence must be sent so the decoding radio can do the reverse - decode the DTMF into T9 text with the extended DTMF decoder. Also the DTMF input mode (* key) sends the radio's DTMF ID by default before the DTMF sequence you have typed, so that needs to be stripped too.

OneOfEleven, 2023-09-19:
The DTMF decoder in the radio currently doesn't decode non-QS radios very well, there are 16 DTMF coefficient values sent to the chip which I think need tuning a bit to fix the problem, but without the detailed register info it could be difficult to work out :(

The DTMF decoder works well with other QS radios though. Looking at the TX DTMF audio sent from the QS radios and the non-QS radios looks OK (and sounds OK), the radio chips DTMF decoder is being far too fussy though - at the moment.

If that can be sorted, I'll look at adding T9 for you.

LZ2DMV, 2023-09-19:
Yes, of course.

Another use case of the T9 implementation might be to be able to input channel names from the radio, without the need of a programming cable, computer and software.

OneOfEleven, 2023-09-19:
Adding a channel name edit menu is something I wanted to add, just haven't got there yet.

closed 2023-10-07 by LZ2DMV