OneOfEleven/uv-k5-firmware-custom #11: [Proposal] have you consider adding voltage dispaly in status bar

zhaoparrot, 2023-09-19:
there is a interesting function at the bottom of this code ↓
https://github.com/piotr022/UV_K5_playground/blob/main/libs/views/rssi_sbar.hpp
it can display battery voltage in the status bar beside the battery icon
I have try to integrate this in your code and bulid my own firmware ,but I'm just not good enough in coding skills.

OneOfEleven, 2023-09-19:
Yes I've thought about adding some battery text (either voltage or percent) to the status bar, but to do so means removing other indicators to make room for the text, such as the keylock and USB-C/charge indicator.

The code you're pointing too shows the battery voltage only when the radio is not on charge and the function key is not pressed and the keylock is not active.

I'll do that and see how it goes, I'll make it a compile option. Bare with me.

OneOfEleven, 2023-09-19:
I've added two compile options, you can now show the battery voltage or the battery percentage on the top status bar. Think I prefer percentage myself. Anyway, let me know if any problems with it.

zhaoparrot, 2023-09-19:
nice work . 
I got two small problems
1.Its update frequency is too slow. thus some case it shows different voltage than the menu voltage. I think a few second should be reasonable.
Same problem goes with the charging percent display.
[image]
2.When i plugin the programming cable it shows a high voltage, that's no problem but the display was missing a pixel. this confuse me...  a 9 is not any fatter than a 8 ,why is this happening
[image]
compare the two pic I uploaded you'll see what am talking about.

OneOfEleven, 2023-09-19:
That is a very high voltage for a lithium battery, keep an eye on your battery !!

Trying to think what would cause that pixel problem. It's a strange one as all displays as far as I know are a fixed 128 pixels wide on all the K5/K6's. Won't stop looking till I work it out.

Is your radio an early version do you know ? .. ie, did you get it when the radio first came out.

OneOfEleven, 2023-09-19:
I've set the update rate for the status bar to a regular 1 second if the user has enabled voltage/percentage in the status bar. Will see how that goes.

OneOfEleven, 2023-09-19:
I've found the pixel problem, hopefully now fixed - let me know.

ANTodorov, 2023-09-19:
I also have the issue with the voltage jump when the programming cable is plugged in. But I think it is QS/design issue of the device because it is available with the original firmware too. 
Procedure to observe it: 
1.Plug in the USB of the programming cable to a PC.
2. Power-on the device and select the voltage menu to monitor the measure
3. Plug in the programming cable to the device
The displayed voltage will jump to 10+ volts ?!?

It is possible to be the programming cable? I am using "WakeSnow Programming cable" which is an octopus with 8 types of WT connectors. I've measured to power consummation on the USB - there is no significant jump on the current., though...

OneOfEleven, 2023-09-19:
I don't see the voltage jump up on my two K5(8)'s. I'm using a baofeng programming cable.

Maybe they changed the schematic slightly with the K5(8) / K6 ?

ANTodorov, 2023-09-19:
I am playing with the "jenuine" UV-K5. Both have same same Voltage behavior...

OneOfEleven, 2023-09-19:
Looks the same as my uv-k5(8)'s.

I'm sure it's not a problem, it's probably something to do with where on the circuit they are monitoring the voltage from using the ADC. I was limiting the voltage reading to 8.45V until today, which was hiding the true reading really, now I just show whatever the voltage is reported by the ADC.

LZ2DMV, 2023-09-19:
Could you also print 'V' after the voltage digits?

OneOfEleven, 2023-09-19:
> Could you also print 'V' after the voltage digits?

No because there's no more room, it would cover up the VOX symbol if I go any further left ..

[image]

LZ2DMV, 2023-09-19:
Maybe you can just check how much real estate is there in the status bar, and if it is full, either do not show voltage at all or show only digits. If there is enough space (almost no icons showing) in the right corner, show ths full thing (digits + V). If some of the space is empty but it is not in the right corner, find where it is and draw it there.

OneOfEleven, 2023-09-19:
Try it now, I've made it work as you suggested.

OneOfEleven, 2023-09-19:
I've added a battery text menu option "BATTXT", so removed the compile option.

If there is space on the status bar, it will show the text, if not enough space, it won't show the text.

LZ2DMV, 2023-09-19:
Works very good!

zhaoparrot, 2023-09-20:
I researched for a while and found the real reason why the 9.30 character occasionally lacked a pixel. The culprit was the battery icon. When the battery changed from 3 to 4 grids, the battery icon would quietly move left and right by one pixel! The latest firmware still has this problem, I will try to analyze the bitmap to see if there is a bug.
As for the problem of excessive voltage, I used a USB ammeter to test the current of the USB interface. It was only about 5mA, which is basically the same as when the flash cable is unloaded. I also tested the battery voltage. After plugging in the programming cable, when the screen showed the voltage rising, there was no fluctuation in the actual battery voltage. My hardware version is V1.4, produced in July 2023.

zhaoparrot, 2023-09-20:
I think this may have something todo with issue#12

OneOfEleven, 2023-09-20:
I think I found it. The code is a bit messy with some things like drawing to the screen from vastly different parts of the firmware, it needs bringing together a bit really to unify some things so as not to cause this type of problem.

I didn't know there was another battery symbol drawing function elsewhere, but I do now.

Please let me know if the problem still happena.

zhaoparrot, 2023-09-20:
Nice job!
I also trackdown the code and did found UI_DisplayBattery,but when i see UI_DisplayBattery(gLowBatteryCountdown);I assume this was only called when the battery is low ,so missed the real reason.

OneOfEleven, 2023-09-20:
I'm assuming the problem showed up when your battery was low, another routine elsewhere in the firmware writes it's own battery symbol when the battery falls below a set level.

zhaoparrot, 2023-09-21:
problem also showed up when i plugin programming cable the battery goes from 3 bar to 4 bar

OneOfEleven, 2023-09-21:
Is it fixed ? .. no ones mentioned if it'[s OK or not now

zhaoparrot, 2023-09-21:
it's fixed now   I mean it also showed up 3 to 4 before you last fix to this , sorry for the misleading.

closed 2023-09-21 by zhaoparrot

OneOfEleven, 2023-09-21:
No no it's fine, just didn't know if the fix worked ok for everyone.