OneOfEleven/uv-k5-firmware-custom #13: can u elaborate on 'COMPANDER' and 'F-CALI'

zhaoparrot, 2023-09-19:
having trouble understand what was those for
F-CALI seems to be adjusting xtal calibration, but why is this need user calibration? is there a hardware problem you found?I see the code is writing some value to bk reg ,but can't figure out what was that doing.
what was defult setting for this calibration(for most users)?

OneOfEleven, 2023-09-19:
The compander is an option QS don't use but it's an option that can be enabled in the radio chip, so I'm adding support for it in the firmware. Companding is when the TX audio is compressed and   the RX audio is decompressed, it's normally used for slightly improving dynamic range, but it can be used to boost the TX audio level (mic gain) significantly.

The F-CALI option is simply to allow the user to finely calibrate the radios TX/RX frequency. The frequency calibration is one of the variables found in the eeprom calibration data. The user needs an accurate frequency counter to adjust it though, it's a normal aspect of ham radio.

OneOfEleven, 2023-09-19:
The default frequency calibration setting varies from radio to radio, which is why Quansheng have included it in the eeprom calibration data. It's just one of many variables that can be accessed, their are also TX power calibration settings, RX signal level calibration settings, quite a few really.

LZ2DMV, 2023-09-19:
It will be useful if you can expose the other calibration settings, maybe in a separate menu. Perhaps it is a good idea to make it hidden or extend the F-LOCK menu, because it is accessible only when booting while holding specific keys.

zhaoparrot, 2023-09-20:
Thanks for your detailed explanation, it's very helpful. Speaking of wireless chips, I saw in an application document that it supports MDC1200, but there is not enough information for further research. Do you think it is feasible to let UV-K5 recognize and display MDC1200 signaling?
[image]

translate ↓

MDC1200
Turn on MDC mode and use RF_EnterMdc(), which will not affect normal audio listening and can be entered at the same time.
DTMF/SELCALL mode for reception. The MDC rate odd register is multiplexed with the Tone2 register. The default mode is
1200/1800, if you want to use 1200/2400 mode, you need to enable macro definition MDC240100
2.Exit using RF_ExitMdc()
3.Use RF_MdcTransmit() to transmit MDC, returning 1 for failure and 0 for success.
4. Use RF_MdcReceive() to receive MDC, returning 1 for failure and 0 for success.
5. Can support HDC1200 mode

pdf file:https://github.com/amnemonic/Quansheng_UV-K5_Firmware/blob/main/hardware/BK4819(V3)%20Application%20Note%2020210428.pdf

OneOfEleven, 2023-09-20:
I did want to do add some FSK stuff, I'll look at it when I get to it, though I have to be careful npt to run out of flash space.

DE8MSH, 2023-09-20:
Can one explain what compander mode is? On TX and/or RX?

closed 2023-09-20 by zhaoparrot