OneOfEleven/uv-k5-firmware-custom #15: SC-REV option not persistant. Reverts back to TO after power cycle.

spamsool, 2023-09-19:
Hi,

I have built 30d6f3b version and noticed that Menu->24 SC-REV option is not persistent. Whatever I change it to, it remains set only until I power off the radio. After Power-On it always reverts to TO setting.
I have also tried official DualTachyon firmware 2023-09-16-no-swd and SC-REV option does remain after reboot.
Could you please check this?

OneOfEleven, 2023-09-19:
I just tried the SC-REV setting and it's remembering it OK here, try the latest version, let me know if it still has a problem.

spamsool, 2023-09-19:
I built the latest firmware with different Makefile options and noticed that that the problem only exists when FM radio is enabled in Makefile. Attempt 1 : I build the firmware using defaults from the Makefile - SC-REV setting is permanent after restart of the radio. Attempt 2: I build the firmware with all the defaults in Makefile except of NABLE_FMRADIO                := 1 -> after flashing this file SC-REV always reverts back to TO after reboot of the radio.

OneOfEleven, 2023-09-19:
You're right.  just re-enabling the FM radio causes it.

I'll be back !

OneOfEleven, 2023-09-19:
I've found and fixed it. I'll upload it after I'm done with what I'm doing at mo.

Thank you for letting me know.

closed 2023-09-20 by OneOfEleven