OneOfEleven/uv-k5-firmware-custom #16: VFO display issue

M7OCM, 2023-09-20:
Hi 111 latest build - display related issue when entering frequency in VFO mode. As the last digit is entered the frequency appears to shift a pixel or 2 to the right. Its barely noticeable but once you see it you can't unsee it ha no big deal but possibly fix worthy? Both VFOs suffer from it.

M7OCM, 2023-09-20:
Should mention that is happening with big digits

zhaoparrot, 2023-09-20:
Now there is one more person can't unsee it 😂

M7OCM, 2023-09-20:
Thanks 111, unfortunately last few commits are generating this:

arm-none-eabi-gcc -Os -Wall -Werror -mcpu=cortex-m0 -fno-builtin -fshort-enums -fno-delete-null-pointer-checks -std=c11 -MMD -DPRINTF_INCLUDE_CONFIG_H -DGIT_HASH=\"\" -DENABLE_AIRCOPY -DENABLE_FMRADIO -DENABLE_OVERLAY -DENABLE_UART -DENABLE_BIG_FREQ -DENABLE_NOAA -DENABLE_VOICE -DENABLE_ALARM -DKEEP_MEM_NAME -DENABLE_WIDE_RX -DENABLE_CTCSS_TAIL_PHASE_SHIFT -DENABLE_MAIN_KEY_HOLD -DENABLE_BOOT_BEEPS -DENABLE_COMPANDER  -I C:/GnuWin32/uv-k5-firmware-custom-main/ -I C:/GnuWin32/uv-k5-firmware-custom-main//external/CMSIS_5/CMSIS/Core/Include/ -I C:/GnuWin32/uv-k5-firmware-custom-main//external/CMSIS_5/Device/ARM/ARMCM0/Include -c driver/bk1080.c -o driver/bk1080.o
In file included from driver/bk1080.c:22:
C:/GnuWin32/uv-k5-firmware-custom-main/misc.h:45:2: error: expected ',' or '}' before 'LAST_CHANNEL'
   45 |  LAST_CHANNEL
      |  ^~~~~~~~~~~~
make: *** [driver/bk1080.o] Error 1

C:\GnuWin32\uv-k5-firmware-custom-main>del /S /Q *.o  1>nul 2>nul

C:\GnuWin32\uv-k5-firmware-custom-main>del /S /Q *.d  1>nul 2>nul

C:\GnuWin32\uv-k5-firmware-custom-main>fw-pack.py firmware.bin 230920 firmware.packed.bin
Traceback (most recent call last):
  File "C:\GnuWin32\uv-k5-firmware-custom-main\fw-pack.py", line 23, in <module>
    plain = open(sys.argv[1], 'rb').read()
            ^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'firmware.bin'

C:\GnuWin32\uv-k5-firmware-custom-main>pause
Press any key to continue . . .

Retried a working version 0b676d2 and that compiles okay.

OneOfEleven, 2023-09-20:
I'm guessing to enable NOAA ?

If so, I found the problem.

zhaoparrot, 2023-09-20:
I tested the latest code ,it works on my end .the vfo issue seems solved.(not enableing noaa)

M7OCM, 2023-09-20:
Believe I did have NOAA in the mix purely for testing! I'll try it again shortly and report back don't need NOAA but good to get bugs figured out :-)

M7OCM, 2023-09-20:
Compilation worked thanks

closed 2023-09-20 by OneOfEleven