OneOfEleven/uv-k5-firmware-custom #17: Menu 20: No EXIT at OFF

DE8MSH, 2023-09-20:
Hi 1O11.

if I set BACKL to OFF and want to exit it won't let me exiting. But I like to exit again ;) Values above OFF working.

OneOfEleven, 2023-09-20:
Couldn't find my way back to the backlight menu option after turning it off on mine lol.

Anyway, hopefully fixed. Let me knoiw.

DE8MSH, 2023-09-20:
Use a flashlight to get back. LOL. Tnx. I'll try...

DE8MSH, 2023-09-20:
Hello 1O11. Works like a charm again. :)

DE8MSH, 2023-09-20:
Sorry. I've to reopening it. EXIT on light OFF doesn't work.

DE8MSH, 2023-09-20:
Did you do a re-do commit here? A few 10 minutes ago it worked?!

OneOfEleven, 2023-09-20:
It seems only to do it under certain sequences, looking for the bug now.

OneOfEleven, 2023-09-20:
Right, I found it this time.

Plus I've made the backlight come on when you go the backlight menu no matter what it's set to - so the user can see what they are doing (I couldn't see any text due to low light level in the room)  !

DE8MSH, 2023-09-20:
Now it stays on >=10 seconds at OFF.

DE8MSH, 2023-09-20:
...until I reach BLIGHT menu??? hahahaha then its turning on?!

OneOfEleven, 2023-09-20:
Only while your setting/changing the menu option, once the light goes off it won't come back on, until you visit the backlight menu again.

I've done that because the screen can't be seen in low light conditions, so finding the backlight menu is possible if the light comes on for that 10 seconds when you cross the backlight menu.

DE8MSH, 2023-09-20:
Got it. It's more intelligent Light management if light stays on whole time at menu? and if you quit it: turn light off immediately if menu quitting with exit button? ( in later version) ;)

DE8MSH, 2023-09-20:
but: no exit at LIGHT OFF is closed now.

closed 2023-09-20 by DE8MSH