OneOfEleven/uv-k5-firmware-custom #19: Just a question....

DE8MSH, 2023-09-20:
Hi 1O11.
I reorganized menu list. 'cause I want more used menu entries to upper list numbers.

menu.c: 

`   const char MenuList[][7] =
{
	"SQL",
    "STEP",
    "T-PWR",     // was "TXP"
    "BACKLT",    // was "ABR"
	"SLIST",

	"S-ADD1",
	"S-ADD2",

	"SLIST1",
	"SLIST2",

`

menu.h:

`enum
{
	MENU_SQL = 0,
	MENU_STEP,
	MENU_TXP,
	MENU_ABR,
	MENU_S_LIST,

	MENU_S_ADD1,
	MENU_S_ADD2,
	MENU_SLIST1,
	MENU_SLIST2,

	MENU_R_DCS,`

Works on setting screen (right side). But I'm missing one thing (again). Menus (left side) are the same. But setting screen shows new structure. What ima missing here? I knew what file had to edit, too. But I forgot which it was... I did that weeks a go for my private joy.

Thank you for lettin me stealing your preacous time ;)

DE8MSH, 2023-09-20:
Wait... It works now. Without a change. I simply didnt do nothing. Sorry for spamming around....

closed 2023-09-20 by DE8MSH

OneOfEleven, 2023-09-20:
You MUST keep the 7 hidden menu items at the end of the lists though - for now