OneOfEleven/uv-k5-firmware-custom #20: Side Key functionality

M7OCM, 2023-09-20:
Bit of a question/idea.

I use side key 1 for Monitor and side key 2 for Scan on/off. Unfortunately Quansheng forgot the off bit! Press SK2, sure enough it scans, press it again it does zilch. Wonder if thats fixable?

While on the subject of scanning and I'm guessing this is far more of a reach. But unless the scan stops on a carrier, the mem channel always reverts to the initial starting point when manually stopped with Exit. This is quite unusual even for Chinese HTs - any ideas?

By the way no need to apologise for bugs, pretty sure we are all thankful for your efforts and afterall we are your beta testers haha

OneOfEleven, 2023-09-20:
I had that problem too. Try it now, I've made a side key and scanner change, hopefully not broke either.

OneOfEleven, 2023-09-20:
> While on the subject of scanning and I'm guessing this is far more of a reach. But unless the scan stops on a carrier, the mem channel always reverts to the initial starting point when manually stopped with Exit. This is quite unusual even for Chinese HTs - any ideas?

I like that actually, but it depends when you press the exit key I think. If you press the exit key while it's stopped on a RX'ed signal it will remain on that channel/freq, but if you press the exit key while it's scanning it reverts back to the original channel/freq.
 
> By the way no need to apologise for bugs, pretty sure we are all thankful for your efforts and afterall we are your beta testers haha

<cracks the whip>

M7OCM, 2023-09-20:
> I had that problem too. Try it now, I've made a side key and scanner change, hopefully not broke either.

Wow you are super quick at fixes, lunch finished and a proper off button to return to, thank you very much.  So far so good not noticed anything awry you'll be pleased to know ;-)