OneOfEleven/uv-k5-firmware-custom #24: Add some detail info about compile directly in windows

zhaoparrot, 2023-09-21:
I followed your instruction and encounterd two issue, consider add some info in readme for other people learning how to do this.
1. after install gcc. I need to manualy add gcc path to my OS environment val PATH like this:
C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin
other wise I got some error.
2. I have to change win_make.bat from this 
fw-pack.py firmware.bin 230920 firmware.packed.bin
to 
**python** fw-pack.py firmware.bin 230920 firmware.packed.bin
It seems that I have encounter this problem few years ago when I first start palying with python. I may have solved this issue before ,but now i forgot how to .....and i got lazy to look it up ,just add 'python' and it will work.

OneOfEleven, 2023-09-21:
Thank you for letting me know. I shall do that shortly,.

I'm just seeing if I can get the makefile to work properly for me when letting it call the python line itself (in windows), windows doesn't have the shell commands that nix has.

OneOfEleven, 2023-09-21:
Need to keep this open till I done it for you otherwise I'll forget all about it :)

OneOfEleven, 2023-09-21:
I've fixed the makefile error I did so it should now call the python line from there rather than having to call it in the .bat file.

zhaoparrot, 2023-09-21:
got a harmless error  : Error9009
works like a charm . 
nice job

```
i/ui.o ui/welcome.o version.o main.o -o firmware
arm-none-eabi-objcopy -O binary firmware firmware.bin
python fw-pack.py firmware.bin 835378c firmware.packed.bin
python3 fw-pack.py firmware.bin 835378c firmware.packed.bin
make: [all] Error9009 (ignore)
arm-none-eabi-size firmware
   text    data     bss     dec     hex filename
  55432       0    2696   58128    e310 firmware

D:\WalkyTalky\CODE\uv-k5-firmware-by1O11>del /S /Q *.o  1>nul 2>nul

D:\WalkyTalky\CODE\uv-k5-firmware-by1O11>del /S /Q *.d  1>nul 2>nul

D:\WalkyTalky\CODE\uv-k5-firmware-by1O11>pause
press any key to continue. . .
```

zhaoparrot, 2023-09-21:
find a typo in win_make.bat
company name is QuanSheng not QuangSheng

OneOfEleven, 2023-09-21:
oh good !

Yes the error doesn't matter, it doesn't know whether to call 'python' or 'python3' so it tries both.

closed 2023-09-21 by OneOfEleven