OneOfEleven/uv-k5-firmware-custom #25: a question about FREQ COPY function

zhaoparrot, 2023-09-21:
I was trying to implement [this](https://github.com/amnemonic/Quansheng_UV-K5_Firmware/blob/main/uvmod_kitchen/mod_disable_freq_copy_timeout.py) 
to make the FREQ COPY function ([F]+4)  run indefinetely or if user press [Exit] button
It seems I can achive this by block this two line

```
	if (gScreenToDisplay == DISPLAY_SCANNER && gScannerEditState == 0 && gScanCssState < SCAN_CSS_STATE_FOUND)
	{
		if (++gScanProgressIndicator > 32)
		{
			if (gScanCssState == SCAN_CSS_STATE_SCANNING && !gScanSingleFrequency)
				gScanCssState = SCAN_CSS_STATE_FOUND;
		/*	else
				gScanCssState = SCAN_CSS_STATE_FAILED;*/

			gUpdateStatus = true;
		}

		gUpdateDisplay = true;
	}
```
not sure if this is the right way of doing it ,seems works for me
then I got a problem : after scan complete ,the UI of scan result will automatic jump back after about 16second ,I would like the scan result to be there forever until I press Exit or M key, Chould u please help me point out where I can control this?

M7OCM, 2023-09-21:
OT No need to use F anymore just hold until it changes. A VERY cool feature!!

The only exception is switching to stored FM radio stations. F+1 (1 being used for band changes in normal use).

OneOfEleven, 2023-09-21:
Bare with me, busy day today, got to go out again shortly then roof guy coming round.

zhaoparrot, 2023-09-22:
Do not worry. I'm already very grateful that you can share this project. Respect.

Meanwhile, I'm researching another issue , in Dualwatch mode. When a signal ends, I only have about 3 seconds to look at the screen and confirm which channel the signal came from (there are two popular relay stations locally). I want to make the icon display for a longer time or continuously display until the next signal comes in. I saw how BITMAP_VFO_NotDefault is displayed on the screen in UI_DisplayMain, but I haven't found how it is erased from the screen.I'm checking APP_TimeSlice500ms now.

OneOfEleven, 2023-09-22:
The UI_DisplayMain() function in ui/main.c draws most all the screen (but not the top status bar).

I don't know all the code myself, I have to do a lot of searches while I'm coding with it to find what I need all the time.

zhaoparrot, 2023-09-22:
thanks
by the way
it seems u r planing to implement a spectrum .  Are u planing to code your own spectrum  or gonna merge fagci's spectrum  
 code?(I don't know if there is a copyright issue or not)

OneOfEleven, 2023-09-22:
To be honest, I'm only really doing this for myself, I only share what I do if people are interested and good about it all, the second ANY problem pops up I tend to stop sharing any of it and leave people to do their own thing, so that I can do my own thing with whatever it is I'm interested in at the time.

zhaoparrot, 2023-09-22:
Thank u for shareing this project ,it's very helpful. Apology for the extra questions I asked. I'll see what I can do.
Please don't stop sharing this .

closed 2023-09-22 by zhaoparrot