OneOfEleven/uv-k5-firmware-custom #26: Menu errors 653a00b

M7OCM, 2023-09-21:
653a00b

Fun times heehee

-c ui/menu.c -o ui/menu.o
ui/menu.c:43:1: error: missing braces around initializer [-Werror=missing-braces]
   43 | {
      | ^
......
   88 |  "AL-MOD",  0, VOICE_ID_INVALID                      },
      |  {                                                   }
ui/menu.c:90:2: error: expected identifier or '(' before '{' token
   90 |  {"ANI-ID", 0, VOICE_ID_ANI_CODE                     },
      |  ^
ui/menu.c:90:55: error: expected identifier or '(' before ',' token
   90 |  {"ANI-ID", 0, VOICE_ID_ANI_CODE                     },
      |                                                       ^
ui/menu.c:91:55: error: expected identifier or '(' before ',' token
   91 |  {"UPCODE", 0, VOICE_ID_INVALID                      },
      |                                                       ^
ui/menu.c:92:55: error: expected identifier or '(' before ',' token
   92 |  {"DWCODE", 0, VOICE_ID_INVALID                      },
      |                                                       ^
ui/menu.c:93:55: error: expected identifier or '(' before ',' token
   93 |  {"D-ST",   0, VOICE_ID_INVALID                      },
      |                                                       ^
ui/menu.c:94:58: error: expected identifier or '(' before ',' token
   94 |     {"D-RSP",  0, VOICE_ID_INVALID                      },
      |                                                          ^
ui/menu.c:95:55: error: expected identifier or '(' before ',' token
   95 |  {"D-HOLD", 0, VOICE_ID_INVALID                      },
      |                                                       ^
ui/menu.c:96:55: error: expected identifier or '(' before ',' token
   96 |  {"D-PRE",  0, VOICE_ID_INVALID                      },
      |                                                       ^
ui/menu.c:97:55: error: expected identifier or '(' before ',' token
   97 |  {"PTT-ID", 0, VOICE_ID_INVALID                      },
      |                                                       ^
ui/menu.c:98:55: error: expected identifier or '(' before ',' token
   98 |  {"D-DCD",  0, VOICE_ID_INVALID                      },
      |                                                       ^
ui/menu.c:99:55: error: expected identifier or '(' before ',' token
   99 |  {"D-LIST", 0, VOICE_ID_INVALID                      },
      |                                                       ^
ui/menu.c:100:55: error: expected identifier or '(' before ',' token
  100 |  {"D-LIVE", 0, VOICE_ID_INVALID                      }, // live DTMF decoder
      |                                                       ^
ui/menu.c:101:55: error: expected identifier or '(' before ',' token
  101 |  {"PONMSG", 0, VOICE_ID_INVALID                      },
      |                                                       ^
ui/menu.c:102:55: error: expected identifier or '(' before ',' token
  102 |  {"ROGER",  0, VOICE_ID_INVALID                      },
      |                                                       ^
ui/menu.c:103:55: error: expected identifier or '(' before ',' token
  103 |  {"BATVOL", 0, VOICE_ID_INVALID                      }, // was "VOL"
      |                                                       ^
ui/menu.c:104:55: error: expected identifier or '(' before ',' token
  104 |  {"BATTXT", 0, VOICE_ID_INVALID                      },
      |                                                       ^
ui/menu.c:105:55: error: expected identifier or '(' before ',' token
  105 |  {"MODE",   0, VOICE_ID_INVALID                      }, // was "AM"
      |                                                       ^
ui/menu.c:107:55: error: expected identifier or '(' before ',' token
  107 |  {"NOAA-S", 0, VOICE_ID_INVALID                      },
      |                                                       ^
ui/menu.c:109:55: error: expected identifier or '(' before ',' token
  109 |  {"RESET",  0, VOICE_ID_INITIALISATION               }, // might be better to move this to the hidden menu items ?
      |                                                       ^
ui/menu.c:114:55: error: expected identifier or '(' before ',' token
  114 |  {"F-LOCK", 1, VOICE_ID_INVALID                      },
      |                                                       ^
ui/menu.c:115:55: error: expected identifier or '(' before ',' token
  115 |  {"TX-200", 1, VOICE_ID_INVALID                      }, // was "200TX"
      |                                                       ^
ui/menu.c:116:55: error: expected identifier or '(' before ',' token
  116 |  {"TX-350", 1, VOICE_ID_INVALID                      }, // was "350TX"
      |                                                       ^
ui/menu.c:117:55: error: expected identifier or '(' before ',' token
  117 |  {"TX-500", 1, VOICE_ID_INVALID                      }, // was "500TX"
      |                                                       ^
ui/menu.c:118:55: error: expected identifier or '(' before ',' token
  118 |  {"350-EN", 1, VOICE_ID_INVALID                      }, // was "350EN"
      |                                                       ^
ui/menu.c:119:55: error: expected identifier or '(' before ',' token
  119 |  {"SCR-EN", 1, VOICE_ID_INVALID                      }, // was "SCREN"
      |                                                       ^
ui/menu.c:121:55: error: expected identifier or '(' before ',' token
  121 |  {"TX-EN",  1, VOICE_ID_INVALID                      }, // enable TX
      |                                                       ^
ui/menu.c:122:55: error: expected identifier or '(' before ',' token
  122 |  {"F-CALI", 1, VOICE_ID_INVALID                      }, // reference xtal calibration
      |                                                       ^
ui/menu.c:125:1: error: expected identifier or '(' before '}' token
  125 | };
      | ^
cc1.exe: all warnings being treated as errors
make: *** [ui/menu.o] Error 1

C:\GnuWin32\uv-k5-firmware-custom-main>del /S /Q *.o  1>nul 2>nul

C:\GnuWin32\uv-k5-firmware-custom-main>del /S /Q *.d  1>nul 2>nul

C:\GnuWin32\uv-k5-firmware-custom-main>pause
Press any key to continue . . .

OneOfEleven, 2023-09-21:
can you copy/paste that first line here ?  (ui/menu.c line 43)

M7OCM, 2023-09-21:
Yep just need to head back to PC

M7OCM, 2023-09-21:
Copied a wedge lol

arm-none-eabi-gcc -Os -Wall -Werror -mcpu=cortex-m0 -fno-builtin -fshort-enums -fno-delete-null-pointer-checks -std=c11 -MMD -DPRINTF_INCLUDE_CONFIG_H -DGIT_HASH=\"\" -DENABLE_AIRCOPY -DENABLE_FMRADIO -DENABLE_OVERLAY -DENABLE_UART -DENABLE_BIG_FREQ -DENABLE_NOAA -DENABLE_VOICE -DENABLE_ALARM -DKEEP_MEM_NAME -DENABLE_WIDE_RX -DENABLE_CTCSS_TAIL_PHASE_SHIFT -DENABLE_MAIN_KEY_HOLD -DENABLE_BOOT_BEEPS -DENABLE_COMPANDER -DENABLE_SHOW_CHARGE_LEVEL -DENABLE_AUDIO_BAR  -I C:/GnuWin32/uv-k5-firmware-custom-main/ -I C:/GnuWin32/uv-k5-firmware-custom-main//external/CMSIS_5/CMSIS/Core/Include/ -I C:/GnuWin32/uv-k5-firmware-custom-main//external/CMSIS_5/Device/ARM/ARMCM0/Include -c ui/menu.c -o ui/menu.o
ui/menu.c:43:1: error: missing braces around initializer [-Werror=missing-braces]
   43 | {
      | ^
......

OneOfEleven, 2023-09-21:
nearly ;)  .. i should have said from the source code file, not the errror :)

M7OCM, 2023-09-21:
Duh sorry line 43 from menu.c in the ui directory heehee will do!

M7OCM, 2023-09-21:
42 const t_menu_item MenuList[] =
43 {
44	{"SQL",    0, VOICE_ID_SQUELCH                      },

Hope thats what you wanted - noob at large!

OneOfEleven, 2023-09-21:
hmm, looks ok.

I don't get the compile problem, trying to work it out

OneOfEleven, 2023-09-21:
Try the new upload I done, though I can't see why you have that compile problem.

M7OCM, 2023-09-21:
Success thanks new file compiled fine possibly just a glitch my end?

closed 2023-09-21 by M7OCM