OneOfEleven/uv-k5-firmware-custom #27: How to reduce Squelch levels ?

INDIAJULIETVICTOR, 2023-09-21:
Hi, I am following your excellent work, many people complain that this version of yours does not receive compared to others (piotr022), but I have seen that it often is' only a problem of low sensitivity of the squelch, in the code you should have the possibility to lower the threshold to eliminate this problem.
Thanks and good work.

OneOfEleven, 2023-09-21:
I've restore a setting I was experimenting with - wide mode weak signal bandwidth value, back to original value now.

Let me know if still a problem.

INDIAJULIETVICTOR, 2023-09-21:
thank you very much..

INDIAJULIETVICTOR, 2023-09-21:
I am a Senior Firmware Engineering, i can help to develope the firmware you if you want you can contact on Telegram @Alcap911..

DE8MSH, 2023-09-21:
@INDIAJULIETVICTOR Join the club:

https://t.me/quansheng_uvk5_en_dev

INDIAJULIETVICTOR, 2023-09-21:
i have just tested your firmware compared to (Ver26_rssi78_bar_vol_sq_XSX_V3_5_2.bin) the results are :

1of11  receive level is the same of XSX -136 dB it is equal.

1of11 ->  squelch min level os -115dB    XSX - > -131 dB

1of11 -> TX modulation is very dark       XSX - > is much better.

INDIAJULIETVICTOR, 2023-09-21:
I tried to lower the SQL Values in RADIO.C as follow : 

		// pInfo->SquelchOpenRSSIThresh -= 60;
		// pInfo->SquelchCloseRSSIThresh -= 60;

but there is little difference  -119 instead of -115

OneOfEleven, 2023-09-21:
You also have to play with the Noise and Glitch params ..

   SquelchOpenNoiseThresh
   SquelchCloseGlitchThresh

   SquelchCloseNoiseThresh
   SquelchOpenGlitchThresh

closed 2023-09-22 by OneOfEleven