OneOfEleven/uv-k5-firmware-custom #31: Audio bar glitches

M7OCM, 2023-09-21:
Early days on this one, noticed in VFO mode and changing BW W to N the audio bars appear. Occasionally seen one left most bar pop up randomly.

Cool feature!

Was thinking last night are there any benefits shifting the Bandwidth menu contents W/N to the Mode menu ie AM, NAM, FM, NFM ?

Just an idea. re TG Also please do get a RT-890 their great, got 2, love to see the magic you could  weave in technicolor lol

OneOfEleven, 2023-09-21:
If you blow hard into the mic whilst TX'ing, you can move the bar right across. I need to logify it to make it usable.

Just looking at a fix for this AM problem - manually adjusting the front end gain in real time according to RSSI to stop the AM demodulator overloading.

M7OCM, 2023-09-22:
> Just looking at a fix for this AM problem - manually adjusting the front end gain in real time according to RSSI to stop the AM demodulator overloading.

Nice fix 111, it controls my high gain antenna nicely without overloading well done.

M7OCM, 2023-09-23:
Latest improvements are top notch thanks 111.

closed 2023-09-23 by M7OCM

OneOfEleven, 2023-09-23:
I haven't done any more with the audio bar graph - yet ;)

Currently working on getting the AM fix working 100%