OneOfEleven/uv-k5-firmware-custom #32: Scan list 1 and list 2 has damaged/overlapped fonts

pawol, 2023-09-22:
Scan list 1 and list 2 has damaged/overlapped fonts. Item 34.53 and 35.53 in the menu. Overlapped fonts are visible when defined scan list ha different in width channels names. You have to try to view that list to see the issue. The issue was seen earlier. On today's build 230921 it is still occuring

OneOfEleven, 2023-09-22:
Someone will have to take a photo as I'm not seeing anything wrong when I go to those menu items.

pawol, 2023-09-22:
[image]

pawol, 2023-09-22:
In the above picture it should be PLKR1 name only but we see some overlapped entry

pawol, 2023-09-22:
I found it depends on ently lenght. 5 characters in the name makes overlapping, 6 characters is dispaeyed properly

OneOfEleven, 2023-09-22:
ah, you must have priority channels setup.

Hopefully fixed by making the channel name small font (if need be). Let me know if still a problem or not.

OneOfEleven, 2023-09-22:
Is it fixed for you ?

pawol, 2023-09-22:
I shall check tomorrow

pawol, 2023-09-22:
I hereby confirm that bux is fixed on the build 3fb5817. 
The ticket can be closed.

closed 2023-09-22 by pawol