OneOfEleven/uv-k5-firmware-custom #34: [Proposal] Show dB value while in AM

pdrbsts, 2023-09-23:
I still get distortion when very close to the transmitter.
If you could show the dB value you are using in your lookup table, at least while you are improving on AM reception, we could see where it starts to distort.
Imagine that I get distortion and it reads -10dB. We could try to lower the gain at -10dB and see if it helps.

OneOfEleven, 2023-09-23:
OK, I've added a compile option in Makefile that lets you see a small bit of data on screen. I need to clean it up a bit but it's there to see.

I can add up to another 19dB of gain reduction, I took that out - for now, but can put it back in if people are still getting bad AM.

pdrbsts, 2023-09-23:
Thank you. Will test.

lohtse, 2023-09-23:
> OK, I've added a compile option in Makefile that lets you see a small bit of data on screen. I need to clean it up a bit but it's there to see.
> 
> I can add up to another 19dB of gain reduction, I took that out - for now, but can put it back in if people are still getting bad AM.

Just flashed the latest... you are certainly making great improvements   when we have dual name+frq  it could do with the name being bold so it is easier to see.. extra data is great too see and useful

Main issue for me is the audio is still not as clear as Faqci fw... when I compare them I can hear audio from his a little clearer..Seems softer somehow... BUT having said that yours FW is a massive improvement over those from 3days ago let alone 3 weeks ago.. And I am happy using it... 

Amazing work buddy please don't stop... and don't take to heart anything said negatively ..

OneOfEleven, 2023-09-23:
I've put back the 19dB of front end adjustment for the AM fix so that can cope with stronger signals.

pdrbsts, 2023-09-23:
According to my tests I only get distortion when idx is 1. Any other value gives a clear reception.
I found out that when it starts with idx 1 it doesn't go back, it stays stuck.  But when it starts with say idx 16 and it goes to 1, it gets back.

OneOfEleven, 2023-09-23:
> Just flashed the latest... you are certainly making great improvements when we have dual name+frq it could do with the name being bold so it is easier to see.. extra data is great too see and [...]

That's probably because I'm currently using the full RX bandwidth in AM mode, there is a weak bandwidth setting that the radio can switch too on weak signals, it does exactly that really, it 'softens the audio', it actually cuts the higher audio frequencies down (to reduce HF hiss).
I can go back back to using the weak signal bandwidth OK, I'm just experimenting.

OneOfEleven, 2023-09-23:
> According to my tests I only get distortion when idx is 1. Any other value gives a clear reception. I found out that when it starts with idx 1 it doesn't go back, it stays stuck. But when it starts with say idx 16 and it goes to 1, it gets back. These tests where without these 19dB. Does is affect AM only where idx is 1?

Yes I've added the extra 19dB table entries back, so index 1 (the lowest front end gain) now goes down an extra 19dB .. see how that works.

pdrbsts, 2023-09-23:
Only when idx is 1 there still is the same distortion.

OneOfEleven, 2023-09-23:
I'm now using the full front end adjustment, it's maxed out, it now goes down to -84dB, it won't go any further.

Are you using the radio on a big/external antenna ?

OneOfEleven, 2023-09-23:
I still need to improve the routine to do better adjustment, I won't stop till it's as good as it can get.

pdrbsts, 2023-09-23:
I'm using a portable airband transceiver to test.

OneOfEleven, 2023-09-23:
oh right ! .. you'll have an ENORMOUSE signal from that if you're transmitting anywhere near close the K5.

lohtse, 2023-09-23:
Just flashed the latest as of this comment... I am now getting a weird pulsing inference static  when it receives any trans.  Doesn't happen on previous

OneOfEleven, 2023-09-23:
Yes that's the area I need to work on now, it's the adjustment routine that's 'hunting' the gain setting, I just need to improve it.

lohtse, 2023-09-23:
> Yes that's the area I need to work on now, it's the adjustment routine that's 'hunting' the gain setting, I just need to improve it.

Arrr ok....

Raul-2017, 2023-09-23:
I observed the same "pumping effect" here, just with medium to high level signals, not with weak ones.   I think it is the gain-reduction working in second plane, that somehow interferes with the audio being received, it makes glitches or similar and the audio is cut for some miliseconds, two times per second or so  :-)

pdrbsts, 2023-09-23:
I'm testing without the antenna on the QS.
What's strange is that when it start with idx 1 it distorts and doesn't go back but when it starts with idx 46 it only goes up to 21 and never distorts) even if I touch the QS with the antenna.

OneOfEleven, 2023-09-23:
I need to get the k5 on the signal generator, some of the dB settings that are in the 'datasheet' that's been cobbled together are wrong, I need to determine what the register dB settings really are. I can then create an accurate gain table that will stop the sometimes sudden gain switch being seen.

I may do that later today, will see.

OneOfEleven, 2023-09-23:
> I observed the same "pumping effect" here, just with medium to high level signals, not with weak ones. I think it is the gain-reduction working in second plane, that somehow interferes with the [...]

yes I've just discovered that some of the the stated dB register setting are incorrect, I need to determine what they really are. That will stop the 'pulsing' (incorrect gain switching).

OneOfEleven, 2023-09-23:
What ever the 'datasheet' means by 'LNA-SHORT' is a bit of a mystery, but the dB values stated are incorrect.

lohtse, 2023-09-23:
latest build (VOEFW-9626dbb) is suffering badly from distortion  transmission ... As of now this is is the worst it has been for a while .  I am testing in the same location and same method as I do all fw's  Sometimes the trans will be ok for a second or two then justb can't hear through the distortion and sometimes it will be a few seconds of distortion and than clears up for a few seconds

I see idx  1 -89db 206 - 121 constantly where as before they was mus much lower etc..   hope you manage to confirm issue and resolve it

OneOfEleven, 2023-09-24:
> Just flashed the latest... you are certainly making great improvements when we have dual name+frq it could do with the name being bold so it is easier to see.. extra data is great too see and useful

Adding another font takes up nearly 1k of flash memory :(

[image]

OneOfEleven, 2023-09-24:
Adjusting the front end gain like this I suspect isn't going to be enough to cope with the range of AM signal strengths, the BK chip has a limited adjustment range. The pulsing was mainly due to incorrect listed dB settings in the lna-short register in the.

Might be best to go with Fagci's method of AM demodulation to reduce the AM problem the radio has. Possibly combining Fagci's method and using the usable 60dB range off available gain adjustment would be the best possible fix.

closed 2023-09-24 by OneOfEleven