OneOfEleven/uv-k5-firmware-custom #35: copy of the contents of the memory in the VFO

INDIAJULIETVICTOR, 2023-09-23:
Congratulations again for your excellent results, I wanted to suggest a change that could be made to the firmware, or the possibility' to transfer the parameters of a memory to the VFO (Mem - > VFO) as used on the most important radio, in order to simplify the operational use of the radio.
Thanks again and good job.

Raul-2017, 2023-09-23:
Absolutely!   Mem -> VFO    a classical and super useful function. It would be really good having this!

INDIAJULIETVICTOR, 2023-09-23:
Thank You very Much... :-)

OneOfEleven, 2023-09-23:
I'll do it tomorrow for you.

Chilling out at mo.

INDIAJULIETVICTOR, 2023-09-24:
Thank you very much, take your time and rest, you're doing a splendid job.

OneOfEleven, 2023-09-25:
Only thing I can think of is to add a menu entry to do the channel-to-vfo copy, but that seems a long winded number of key presses to start the copy (user has to find the menu entry to perform the copy).

Adding a keypad function/long-press would be easiest for user to use, but that means sacrificing one of the other function keys.

INDIAJULIETVICTOR, 2023-09-25:
in fact, using a menu item I don't see it practical either, at the moment we have that the F button is almost useless, it is' just a repetition of commands that with long pressure are more practical and faster
therefore it could be used precisely for secondary functions.

OneOfEleven, 2023-09-26:
Yes you're right, the F-key is now defunct really, it could be used as a stand alone quick press  copy Chan to VFO (it still locks/unlocks the keyboard if held pressed).

OneOfEleven, 2023-09-26:
There's also long press MENU key, there currently is no long press function on the M-key, so that also could do the copy.

INDIAJULIETVICTOR, 2023-09-26:
Exactly, the F key now has an interesting use for secondary functions, for example F-3 (VFO / MR) would be excellent because it can be easily memorized

just for example a list of possible secondary functions to be maybe adopted in future changes :
-F 2 (A/B) (Single or Double VFO)
-F 3 (MEM to VFO)
-F * (Scan) for Scanning a specified range of frequency (min --- max)
-F 1 (Band) for activate the Spectrum

etc..

OneOfEleven, 2023-10-03:
I got there at least. I had to learn more about the code before adding the mem-vfo copy.

You have one VFO set to CHANNEL mode, and the other VFO set to VFO mode, then just long press the 'M' key (Menu key), it will copy the channel you see into the VFO you see.

INDIAJULIETVICTOR, 2023-10-03:
please look at this modified code, could you think it would be more optimized in terms of execution speed and / or code ?

`			uint8_t *p_line = p_line1 + LCD_WIDTH;
			switch(Level)
			{
				default:
				case 7:
					memmove(p_line + 20, BITMAP_AntennaLevel6, sizeof(BITMAP_AntennaLevel6));
				case 6:	
					memmove(p_line + 17, BITMAP_AntennaLevel5, sizeof(BITMAP_AntennaLevel5));
				case 5:
					memmove(p_line + 14, BITMAP_AntennaLevel4, sizeof(BITMAP_AntennaLevel4));
				case 4:
					memmove(p_line + 11, BITMAP_AntennaLevel3, sizeof(BITMAP_AntennaLevel3));
				case 3:
					memmove(p_line +  8, BITMAP_AntennaLevel2, sizeof(BITMAP_AntennaLevel2));
				case 2:
					memmove(p_line +  5, BITMAP_AntennaLevel1, sizeof(BITMAP_AntennaLevel1));
				case 1:
					memmove(p_line +  0, BITMAP_Antenna,       sizeof(BITMAP_Antenna));
				case 0:
					break;	
			}`

OneOfEleven, 2023-10-04:
I've done that for you, seems fine.

closed 2023-10-04 by OneOfEleven