OneOfEleven/uv-k5-firmware-custom #38: bad distortion on AM

lohtse, 2023-09-24:
https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/48331155/9d740b36-ff73-4131-862c-acb0d8cc50af

this is you latest build as of the time of this comment

 bad distortion gets worse the longer a trans happens

lohtse, 2023-09-24:
this is the best FW I have so far

https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/48331155/826f9c2e-56c8-42bd-83c4-964639d4bfa9

sorry fr HT low volume that was my fault..

But this is what I compare with(FW)   you has great functiobs etc but that distortion happens and worsens in  most case making the FW unsuable in AM

OneOfEleven, 2023-09-24:
The poor little radio is overloaded with RF there, the front end gain is at it's minimum possible, yet the AM demodulation is still overloaded/saturated.

The front end has no more adjustment range possible to fix that level of RX'ed signal :

A £300++ transceiver will always have a much better dynamic range than a £13 cheap chinese radio.

Fagci's method seems the best way to go I think.

lohtse, 2023-09-24:
I agree that £25(I got when they first started appearing on ebay lol) will never come close to a proper AM system.. But for the money its damn good...     Yeah fagcis is good but the vid I posted is a fw from a few days ago and so far is the best one I have had in all conditions and his newer FW are now starting to go the same way as yours for me.

closed 2023-09-24 by OneOfEleven