OneOfEleven/uv-k5-firmware-custom #39: Bug whith display

Romashca, 2023-09-24:
[image]
[image]
In transmitting On 145 MHz display is  inverted picture or blind

Romashca, 2023-09-24:
when transmitting at 145 MHz, the image on the display begins to fade or invert.

OneOfEleven, 2023-09-24:
Least of your problems by the looks of it lol

Romashca, 2023-09-24:
where can I find your latest release?

OneOfEleven, 2023-09-24:
Damaged radio there.

Same place as usual .. https://github.com/OneOfEleven/uv-k5-firmware-custom

Romashca, 2023-09-24:
I can 't assemble the firmware bin file , too little knowledge, it 's difficult for me, is there a ready-made file ? I would be very grateful

lohtse, 2023-09-24:
> I can 't assemble the firmware bin file , too little knowledge, it 's difficult for me, is there a ready-made file ? I would be very grateful

jus do as it says in readme file it is really that simple

lohtse, 2023-09-24:
[image]

lohtse, 2023-09-24:
you could even just download the file 
[image]

its just there!!

Romashca, 2023-09-24:
[image]

lohtse, 2023-09-24:
issue is your internet

[image]

Romashca, 2023-09-24:
install all .and i dont now what next

lohtse, 2023-09-24:
> install all .and i dont now what next

try actually reading things 

Then you can run 'win_make.bat' from the directory you saved this source code too. You may need to edit the bat file (path to make.exe) depending on where you installed the above two packages too.

I've left some notes in the win_make.bat file to maybe help with stuff.

Romashca, 2023-09-24:
[image]
somthing wrong

OneOfEleven, 2023-09-24:
You're nearly there !

I'm just trying to decipher what's happening.

OneOfEleven, 2023-09-24:
Can you post the test, not an image. I don't understand russian, I need to copy paste in translator.

OneOfEleven, 2023-09-24:
copy and past the error text here.

Romashca, 2023-09-24:
C:\Users\drram\OneDrive\UV K-5\uv-k5-firmware-custom-main>del /S /Q *.o  1>nul 2>nul

C:\Users\drram\OneDrive\UV K-5\uv-k5-firmware-custom-main>del /S /Q *.d  1>nul 2>nul

C:\Users\drram\OneDrive\UV K-5\uv-k5-firmware-custom-main>del /S /Q *.bin  1>nul 2>nul

C:\Users\drram\OneDrive\UV K-5\uv-k5-firmware-custom-main>"C:\Program Files (x86)\GnuWin32\bin\make.exe"
process_begin: CreateProcess(NULL, git rev-parse --short HEAD, ...) failed.
GIT_HASH =
arm-none-eabi-gcc -c -mcpu=cortex-m0 -DENABLE_OVERLAY start.S -o start.o
process_begin: CreateProcess(NULL, arm-none-eabi-gcc -c -mcpu=cortex-m0 -DENABLE_OVERLAY start.S -o start.o, ...) failed.
make (e=2): Не удается найти указанный файл.
make: *** [start.o] Ошибка 2

C:\Users\drram\OneDrive\UV K-5\uv-k5-firmware-custom-main>del /S /Q *.o  1>nul 2>nul

C:\Users\drram\OneDrive\UV K-5\uv-k5-firmware-custom-main>del /S /Q *.d  1>nul 2>nul

C:\Users\drram\OneDrive\UV K-5\uv-k5-firmware-custom-main>pause
Для продолжения нажмите любую клавишу . . .

OneOfEleven, 2023-09-24:
Missing file(s) but not sure which yet.

Do you have 'CMSIS_5' and 'printf' directories in the 'external' directory ?

Romashca, 2023-09-24:
In uv-k5-firmware-custom-main ? or not

OneOfEleven, 2023-09-24:
start.S seems to be missing, but not sure.

Romashca, 2023-09-24:
[image]

OneOfEleven, 2023-09-24:
That looks OK.

I'm trying to think what would cause your error.

OneOfEleven, 2023-09-24:
I wonder if the compiler is not allowed to write to the directory.
What OS is it ?

Romashca, 2023-09-24:
win10

OneOfEleven, 2023-09-24:
It looks like you might have saved the firmware source code intoa directory in the C:\Program files folder. I'm having trouble with the russian.

OneOfEleven, 2023-09-24:
Windows can be fussy about where you have the firmware source code saved too, it might be a windows permissions thing. Not sure yet.

OneOfEleven, 2023-09-24:
Move your **uv-k5-firmware-custom-main** directory to somewhere else that you can have write access to, maybe try it here to start with .. **c:\uv-k5-firmware-custom-main**

Romashca, 2023-09-24:
nothing

OneOfEleven, 2023-09-24:
Does windows let you edit one of the source code files and save it back to the same folder ?

Romashca, 2023-09-24:
yes

OneOfEleven, 2023-09-24:
Out of ideas at the moment, I'll get back to you if I think of something.

zhaoparrot, 2023-09-25:
> C:\Users\drram\OneDrive\UV K-5\uv-k5-firmware-custom-main>del /S /Q *.o 1>nul 2>nul
>
> C:\Users\drram\OneDrive\UV K-5\uv-k5-firmware-custom-main>del /S /Q *.d 1>nul 2>nul
>
> C:\Users\drram\OneDrive\UV [...]

I think I have been there.
make sure you have this in your system path

`C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin`

change it if you installed it somewhere else.
it should be like this 
[image]

OneOfEleven, 2023-09-25:
I also didn't have the path in the OS PATH list, but it's still OK, even though I do mention to add it to your PATH list.

Windows 10 ..

Right click 'This PC', select 'Properties'.
Select 'Advanced System Settings' listed on the right side of the pop-up.
Select 'Environment Variables'.
Edit the 'Path' variable in the lower list box.
Select 'New', enter "C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin" at the bottom.
Click 'OK', then 'OK'.

Might have to reboot windows after that.

The path can be added from the command line as well.

I could also add it into the win_make.bat file I guess (as a temporary path while it compiles).

OneOfEleven, 2023-09-25:
I've updated the win_make.bat file to maybe help with this problem.

Let me know either way, or if I've broken the .bat.

OneOfEleven, 2023-09-25:
> > I've updated the win_make.bat file to maybe help with this problem.
> > Let me know either way, or if I've broken the .bat.
> 
> Incase you need it.
> 
> Test win_make.bat and alls good ... Sae with compiling thought VS studio code(still have to edit makefile in note pad)

Thank you @lohtse !

I was wondering if you'd found a solution to the VSCode problem.

I'd like to set my VSCode up to edit/compile the project but not sure how to set VSCode up. I use it for arduino and other projects (ESP8266 WiFi), so it's ready and waiting to be configured.

lohtse, 2023-09-25:
> Thank you @lohtse !
>
> I was wondering if you'd found a solution to the VSCode problem.
>
> I'd like to set my VSCode up to edit/compile the project but not sure how to set VSCode up. I use it for [...]

The only solition I found was to edit makefile in notpad and everything else is ok in VS studio code... No idea why... Like you I do stuff with adrunio's PI's and CIQ(garmin)  in VS sudio code..  and all works fine, be damned if I know why it will not with makefile lol...   As to setting up vscode not sure if it is any dif to VSSC but would like to think itwas as easy/simple... I just installed the required components etc and it generally works(untill VSSC gets an update lil)

Romashca, 2023-09-25:
C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\arm-none-eabi\uv-k5-firmware-custom-main>del /S /Q *.o  1>nul 2>nul

C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\arm-none-eabi\uv-k5-firmware-custom-main>del /S /Q *.d  1>nul 2>nul

C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\arm-none-eabi\uv-k5-firmware-custom-main>del /S /Q *.bin  1>nul 2>nul

C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\arm-none-eabi\uv-k5-firmware-custom-main>"C:\Program Files (x86)\GnuWin32\bin\make"
process_begin: CreateProcess(NULL, git rev-parse --short HEAD, ...) failed.
GIT_HASH =
arm-none-eabi-gcc -c -mcpu=cortex-m0 -DENABLE_OVERLAY start.S -o start.o
arm-none-eabi-gcc -Os -Wall -Werror -mcpu=cortex-m0 -fno-builtin -fshort-enums -fno-delete-null-pointer-checks -std=c11 -MMD -DPRINTF_INCLUDE_CONFIG_H -DGIT_HASH=\"\" -DENABLE_OVERLAY -DENABLE_UART -DENABLE_SMALL_BOLD -DKEEP_MEM_NAME -DENABLE_WIDE_RX -DENABLE_CTCSS_TAIL_PHASE_SHIFT -DENABLE_MAIN_KEY_HOLD -DENABLE_COMPANDER -DENABLE_REVERSE_BAT_SYMBOL -DENABLE_AM_FIX -DENABLE_AM_FIX_SHOW_DATA  -I C:/ ./ (x86)/ ./ ./ Toolchain/ 2021.10/arm-none-eabi/uv-k5-firmware-custom-main/ -I C:/ ./ (x86)/ ./ ./ Toolchain/ 2021.10/arm-none-eabi/uv-k5-firmware-custom-main//external/CMSIS_5/CMSIS/Core/Include/ -I C:/ ./ (x86)/ ./ ./ Toolchain/ 2021.10/arm-none-eabi/uv-k5-firmware-custom-main//external/CMSIS_5/Device/ARM/ARMCM0/Include -c init.c -o init.o
arm-none-eabi-gcc: error: (x86)/: No such file or directory
arm-none-eabi-gcc: error: Toolchain/: No such file or directory
arm-none-eabi-gcc: error: 2021.10/arm-none-eabi/uv-k5-firmware-custom-main/: No such file or directory
arm-none-eabi-gcc: error: (x86)/: No such file or directory
arm-none-eabi-gcc: error: Toolchain/: No such file or directory
arm-none-eabi-gcc: error: 2021.10/arm-none-eabi/uv-k5-firmware-custom-main//external/CMSIS_5/CMSIS/Core/Include/: No such file or directory
arm-none-eabi-gcc: error: (x86)/: No such file or directory
arm-none-eabi-gcc: error: Toolchain/: No such file or directory
arm-none-eabi-gcc: error: 2021.10/arm-none-eabi/uv-k5-firmware-custom-main//external/CMSIS_5/Device/ARM/ARMCM0/Include: No such file or directory
make: *** [init.o] Ошибка 1

C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\arm-none-eabi\uv-k5-firmware-custom-main>del /S /Q *.o  1>nul 2>nul

C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\arm-none-eabi\uv-k5-firmware-custom-main>del /S /Q *.d  1>nul 2>nul

C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\arm-none-eabi\uv-k5-firmware-custom-main>pause
Для продолжения нажмите любую клавишу . . .

Romashca, 2023-09-25:
C:\uv-k5-firmware-custom-main>del /S /Q *.o  1>nul 2>nul

C:\uv-k5-firmware-custom-main>del /S /Q *.d  1>nul 2>nul 

C:\uv-k5-firmware-custom-main>del /S /Q *.bin  1>nul 2>nul 

C:\uv-k5-firmware-custom-main>"C:\Program Files (x86)\GnuWin32\bin\make"
process_begin: CreateProcess(NULL, git rev-parse --short HEAD, ...) failed.
GIT_HASH =
arm-none-eabi-gcc -c -mcpu=cortex-m0 -DENABLE_OVERLAY start.S -o start.o
arm-none-eabi-gcc -Os -Wall -Werror -mcpu=cortex-m0 -fno-builtin -fshort-enums -fno-delete-null-pointer-checks -std=c11 -MMD -DPRINTF_INCLUDE_CONFIG_H -DGIT_HASH=\"\" -DENABLE_OVERLAY -DENABLE_UART -DENABLE_SMALL_BOLD -DKEEP_MEM_NAME -DENABLE_WIDE_RX -DENABLE_CTCSS_TAIL_PHASE_SHIFT -DENABLE_MAIN_KEY_HOLD -DENABLE_COMPANDER -DENABLE_REVERSE_BAT_SYMBOL -DENABLE_AM_FIX -DENABLE_AM_FIX_SHOW_DATA  -I C:/uv-k5-firmware-custom-main/ -I C:/uv-k5-firmware-custom-main//external/CMSIS_5/CMSIS/Core/Include/ -I 
C:/uv-k5-firmware-custom-main//external/CMSIS_5/Device/ARM/ARMCM0/Include -c init.c -o init.o
arm-none-eabi-gcc -Os -Wall -Werror -mcpu=cortex-m0 -fno-builtin -fshort-enums -fno-delete-null-pointer-checks -std=c11 -MMD -DPRINTF_INCLUDE_CONFIG_H -DGIT_HASH=\"\" -DENABLE_OVERLAY -DENABLE_UART -DENABLE_SMALL_BOLD -DKEEP_MEM_NAME -DENABLE_WIDE_RX -DENABLE_CTCSS_TAIL_PHASE_SHIFT -DENABLE_MAIN_KEY_HOLD -DENABLE_COMPANDER -DENABLE_REVERSE_BAT_SYMBOL -DENABLE_AM_FIX -DENABLE_AM_FIX_SHOW_DATA  -I C:/uv-k5-firmware-custom-main/ -I C:/uv-k5-firmware-custom-main//external/CMSIS_5/CMSIS/Core/Include/ -I 
C:/uv-k5-firmware-custom-main//external/CMSIS_5/Device/ARM/ARMCM0/Include -c sram-overlay.c -o sram-overlay.o
arm-none-eabi-gcc -Os -Wall -Werror -mcpu=cortex-m0 -fno-builtin -fshort-enums -fno-delete-null-pointer-checks -std=c11 -MMD -DPRINTF_INCLUDE_CONFIG_H -DGIT_HASH=\"\" -DENABLE_OVERLAY -DENABLE_UART -DENABLE_SMALL_BOLD -DKEEP_MEM_NAME -DENABLE_WIDE_RX -DENABLE_CTCSS_TAIL_PHASE_SHIFT -DENABLE_MAIN_KEY_HOLD -DENABLE_COMPANDER -DENABLE_REVERSE_BAT_SYMBOL -DENABLE_AM_FIX -DENABLE_AM_FIX_SHOW_DATA  -I C:/uv-k5-firmware-custom-main/ -I C:/uv-k5-firmware-custom-main//external/CMSIS_5/CMSIS/Core/Include/ -I 
C:/uv-k5-firmware-custom-main//external/CMSIS_5/Device/ARM/ARMCM0/Include -c external/printf/printf.c -o external/printf/printf.o
arm-none-eabi-gcc -Os -Wall -Werror -mcpu=cortex-m0 -fno-builtin -fshort-enums -fno-delete-null-pointer-checks -std=c11 -MMD -DPRINTF_INCLUDE_CONFIG_H -DGIT_HASH=\"\" -DENABLE_OVERLAY -DENABLE_UART -DENABLE_SMALL_BOLD -DKEEP_MEM_NAME -DENABLE_WIDE_RX -DENABLE_CTCSS_TAIL_PHASE_SHIFT -DENABLE_MAIN_KEY_HOLD -DENABLE_COMPANDER -DENABLE_REVERSE_BAT_SYMBOL -DENABLE_AM_FIX -DENABLE_AM_FIX_SHOW_DATA  -I C:/uv-k5-firmware-custom-main/ -I C:/uv-k5-firmware-custom-main//external/CMSIS_5/CMSIS/Core/Include/ -I 
C:/uv-k5-firmware-custom-main//external/CMSIS_5/Device/ARM/ARMCM0/Include -c driver/adc.c -o driver/adc.o
arm-none-eabi-gcc -Os -Wall -Werror -mcpu=cortex-m0 -fno-builtin -fshort-enums -fno-delete-null-pointer-checks -std=c11 -MMD -DPRINTF_INCLUDE_CONFIG_H -DGIT_HASH=\"\" -DENABLE_OVERLAY -DENABLE_UART -DENABLE_SMALL_BOLD -DKEEP_MEM_NAME -DENABLE_WIDE_RX -DENABLE_CTCSS_TAIL_PHASE_SHIFT -DENABLE_MAIN_KEY_HOLD -DENABLE_COMPANDER -DENABLE_REVERSE_BAT_SYMBOL -DENABLE_AM_FIX -DENABLE_AM_FIX_SHOW_DATA  -I C:/uv-k5-firmware-custom-main/ -I C:/uv-k5-firmware-custom-main//external/CMSIS_5/CMSIS/Core/Include/ -I 
C:/uv-k5-firmware-custom-main//external/CMSIS_5/Device/ARM/ARMCM0/Include -c driver/aes.c -o driver/aes.o
arm-none-eabi-gcc -Os -Wall -Werror -mcpu=cortex-m0 -fno-builtin -fshort-enums -fno-delete-null-pointer-checks -std=c11 -MMD -DPRINTF_INCLUDE_CONFIG_H -DGIT_HASH=\"\" -DENABLE_OVERLAY -DENABLE_UART -DENABLE_SMALL_BOLD -DKEEP_MEM_NAME -DENABLE_WIDE_RX -DENABLE_CTCSS_TAIL_PHASE_SHIFT -DENABLE_MAIN_KEY_HOLD -DENABLE_COMPANDER -DENABLE_REVERSE_BAT_SYMBOL -DENABLE_AM_FIX -DENABLE_AM_FIX_SHOW_DATA  -I C:/uv-k5-firmware-custom-main/ -I C:/uv-k5-firmware-custom-main//external/CMSIS_5/CMSIS/Core/Include/ -I 
C:/uv-k5-firmware-custom-main//external/CMSIS_5/Device/ARM/ARMCM0/Include -c driver/backlight.c -o driver/backlight.o
arm-none-eabi-gcc -Os -Wall -Werror -mcpu=cortex-m0 -fno-builtin -fshort-enums -fno-delete-null-pointer-checks -std=c11 -MMD -DPRINTF_INCLUDE_CONFIG_H -DGIT_HASH=\"\" -DENABLE_OVERLAY -DENABLE_UART -DENABLE_SMALL_BOLD -DKEEP_MEM_NAME -DENABLE_WIDE_RX -DENABLE_CTCSS_TAIL_PHASE_SHIFT -DENABLE_MAIN_KEY_HOLD -DENABLE_COMPANDER -DENABLE_REVERSE_BAT_SYMBOL -DENABLE_AM_FIX -DENABLE_AM_FIX_SHOW_DATA  -I C:/uv-k5-firmware-custom-main/ -I C:/uv-k5-firmware-custom-main//external/CMSIS_5/CMSIS/Core/Include/ -I 
C:/uv-k5-firmware-custom-main//external/CMSIS_5/Device/ARM/ARMCM0/Include -c driver/bk4819.c -o driver/bk4819.o
arm-none-eabi-gcc -Os -Wall -Werror -mcpu=cortex-m0 -fno-builtin -fshort-enums -fno-delete-null-pointer-checks -std=c11 -MMD -DPRINTF_INCLUDE_CONFIG_H -DGIT_HASH=\"\" -DENABLE_OVERLAY -DENABLE_UART -DENABLE_SMALL_BOLD -DKEEP_MEM_NAME -DENABLE_WIDE_RX -DENABLE_CTCSS_TAIL_PHASE_SHIFT -DENABLE_MAIN_KEY_HOLD -DENABLE_COMPANDER -DENABLE_REVERSE_BAT_SYMBOL -DENABLE_AM_FIX -DENABLE_AM_FIX_SHOW_DATA  -I C:/uv-k5-firmware-custom-main/ -I C:/uv-k5-firmware-custom-main//external/CMSIS_5/CMSIS/Core/Include/ -I 
C:/uv-k5-firmware-custom-main//external/CMSIS_5/Device/ARM/ARMCM0/Include -c driver/crc.c -o driver/crc.o
[... cut ...]
Version suffix is too big!
make: [all] Ошибка 1 (игнорирована)
python3 fw-pack.py firmware.bin  firmware.packed.bin
Version suffix is too big!
make: [all] Ошибка 1 (игнорирована)
arm-none-eabi-size firmware
   text    data     bss     dec     hex filename
  52916       0    2664   55580    d91c firmware

C:\uv-k5-firmware-custom-main>del /S /Q *.o  1>nul 2>nul

C:\uv-k5-firmware-custom-main>del /S /Q *.d  1>nul 2>nul 

C:\uv-k5-firmware-custom-main>pause
Для продолжения нажмите любую клавишу . . .

Romashca, 2023-09-25:
all work !!! Thanks  you

lohtse, 2023-09-25:
> all work !!! Thanks you

Don't forget to use something like notepd to edit the makefile to enable/disable features......

OneOfEleven, 2023-09-25:
I use notepad++ for the text editing.

Romashca, 2023-09-25:
[image]

lohtse, 2023-09-25:
> [image]

Yeah if thats VS studio Code... I use it and regardless of what I do in VSSC it would neve build a FW with  choices I made.. So I have to edit with external editor first for it to work.... The way you will notice if it has is the file size should be bigger than defaulf settings(which is around 54kbs)

so just be aware if a possible issue

Romashca, 2023-09-25:
Whear i cen rename this firmware?

lohtse, 2023-09-25:
> Whear i cen rename this firmware?

in the folder its saved too.. just rename itt!

OneOfEleven, 2023-09-25:
hmmm, seems you've all got vscode working ok with the project. going to see how to use it to build the project.

OneOfEleven, 2023-09-25:
You mean the directory name ?

Romashca, 2023-09-25:
OEFW- *******

OneOfEleven, 2023-09-25:
version.c

Romashca, 2023-09-25:
[image]
Just save changes

Romashca, 2023-09-25:
Version suffix is too big!

OneOfEleven, 2023-09-25:
Try ..

```

#ifdef GIT_HASH
	#define VER     GIT_HASH
#else
	#define VER     "230925"
#endif

//const char Version[]      = "OEFW-"VER;
const char Version[]      = "OEFW-Nurzull";
const char UART_Version[] = "UV-K5 Firmware, Open Edition, OEFW-"VER"\r\n";

```

OneOfEleven, 2023-09-25:
Actually just leave **version.c** as it was, and edit Makefile to be like this ..

```
# the user might not have/want git installed
#GIT_HASH := $(shell git rev-parse --short HEAD)
GIT_HASH := Nerzull
$(info GIT_HASH = $(GIT_HASH)) 
```

Romashca, 2023-09-25:
all done ! Issue was in mic meter when tx

OneOfEleven, 2023-09-25:
Which issue ?

OneOfEleven, 2023-09-25:
If there's a bug you've found please share.

OneOfEleven, 2023-09-25:
Take but don't share eh

closed 2023-09-25 by OneOfEleven