OneOfEleven/uv-k5-firmware-custom #40: Compiling nolonger works

lohtse, 2023-09-24:
just start getting this ...am_fix.c:118:2: error: expected '}' before 'const'
  118 |  const unsigned int original_index = 1;
      |  ^~~~~
am_fix.c:105:2: note: to match this '{'
  105 |  {
      |  ^
am_fix.c:104:35: error: 'am_fix_gain_table' defined but not used [-Werror=unused-const-variable=]
  104 |  static const t_am_fix_gain_table am_fix_gain_table[] =
      |                                   ^~~~~~~~~~~~~~~~~
am_fix.c:100:23: error: 'pga_dB' defined but not used [-Werror=unused-const-variable=]
  100 |  static const int16_t pga_dB[]       = {-33, -27, -21, -15, -9, -6, -3, 0};
      |                       ^~~~~~
am_fix.c:99:23: error: 'mixer_dB' defined but not used [-Werror=unused-const-variable=]
   99 |  static const int16_t mixer_dB[]     = { -8,  -6,  -3,   0};
      |                       ^~~~~~~~
am_fix.c:98:23: error: 'lna_dB' defined but not used [-Werror=unused-const-variable=]
   98 |  static const int16_t lna_dB[]       = {-24, -19, -14,  -9, -6, -4, -2, 0};
      |                       ^~~~~~
am_fix.c:97:23: error: 'lna_short_dB' defined but not used [-Werror=unused-const-variable=]
   97 |  static const int16_t lna_short_dB[] = {-33, -30  -24,   0};   // corrected'ish
      |                       ^~~~~~~~~~~~
cc1.exe: all warnings being treated as errors
make: *** [am_fix.o] Error 1

D:\_______\uv-k5-firmware-custom>del /S /Q *.o  1>nul 2>nul

Been happening since around the last 3 sets of commits on GH

OneOfEleven, 2023-09-24:
Forgot to add '};'

That test option is just a temporary bit if code to let me test the front end registers.

lohtse, 2023-09-24:
> Forgot to add '};'
> 
> That test option is just a temporary bit if code to let me test the front end registers.

Arrr ok no worries

https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/48331155/099e7097-aef7-486d-9fad-8bbf05605343

On the plus side the prebuilt bin file you have uplaoded is now the best I have heard from your fw in some time

lohtse, 2023-09-24:
Thank you for re-adding the compile option... what options do yu have enabled/disabled in the preconfig firmware file on GH??  I compiled and am back to the poor reception/clarity issues have reported..

Used these setup so far

ENABLE_SWD                    := 0
ENABLE_OVERLAY                := 1
ENABLE_UART                   := 1
ENABLE_AIRCOPY                := 0
ENABLE_FMRADIO                := 0
ENABLE_NOAA                   := 0
ENABLE_VOICE                  := 1
ENABLE_ALARM                  := 0
ENABLE_BIG_FREQ               := 0
ENABLE_SMALL_BOLD             := 1
ENABLE_KEEP_MEM_NAME          := 1
ENABLE_WIDE_RX                := 1
ENABLE_TX_WHEN_AM             := 1
ENABLE_CTCSS_TAIL_PHASE_SHIFT := 1
ENABLE_MAIN_KEY_HOLD          := 1
ENABLE_BOOT_BEEPS             := 0
ENABLE_COMPANDER              := 1
ENABLE_SHOW_CHARGE_LEVEL      := 1
ENABLE_REVERSE_BAT_SYMBOL     := 1
ENABLE_AM_FIX                 := 1
ENABLE_AM_FIX_SHOW_DATA       := 1
ENABLE_AM_FIX_TEST1           := 1
ENABLE_SQUELCH1_LOWER         := 1
ENABLE_AUDIO_BAR              := 1
#ENABLE_SINGLE_VFO_CHAN       := 1
#ENABLE_BAND_SCOPE            := 1

and that

ENABLE_SWD                    := 0
ENABLE_OVERLAY                := 1
ENABLE_UART                   := 1
ENABLE_AIRCOPY                := 0
ENABLE_FMRADIO                := 1
ENABLE_NOAA                   := 0
ENABLE_VOICE                  := 1
ENABLE_ALARM                  := 0
ENABLE_BIG_FREQ               := 0
ENABLE_SMALL_BOLD             := 1
ENABLE_KEEP_MEM_NAME          := 1
ENABLE_WIDE_RX                := 1
ENABLE_TX_WHEN_AM             := 1
ENABLE_CTCSS_TAIL_PHASE_SHIFT := 1
ENABLE_MAIN_KEY_HOLD          := 1
ENABLE_BOOT_BEEPS             := 0
ENABLE_COMPANDER              := 1
ENABLE_SHOW_CHARGE_LEVEL      := 1
ENABLE_REVERSE_BAT_SYMBOL     := 1
ENABLE_AM_FIX                 := 1
ENABLE_AM_FIX_SHOW_DATA       := 1
ENABLE_AM_FIX_TEST1           := 1
ENABLE_SQUELCH1_LOWER         := 1
ENABLE_AUDIO_BAR              := 1
#ENABLE_SINGLE_VFO_CHAN       := 1
#ENABLE_BAND_SCOPE            := 1

OneOfEleven, 2023-09-24:
"ENABLE_AM_FIX_TEST1 := 1"

Set that to '0'

lohtse, 2023-09-24:
> "ENABLE_AM_FIX_TEST1 := 1"
> 
> Set that to '0'
ooooo ok

lohtse, 2023-09-24:
OH HELL now thats near damn perfect for me!!!!!!

https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/48331155/06e2320c-2703-4e8c-8d10-0c56b654c18b

OneOfEleven, 2023-09-24:
oh good :)

I found a bug in the code I wrote to create the register look-up table. It was creating incorrect table entries.

It will still fail though with too string a signal, the radio's hardware just can't cope with them.

lohtse, 2023-09-24:
> oh good :)
> 
> I found a bug in the code I wrote to create the register look-up table. It was creating incorrect table entries.
> 
> It will still fail though with too string a signal, the radio's hardware just can't cope with them.

Buddy you have still done wonders...Given the quality and cost of the radio  I don't expect miracles and if the current quality etc I am experiencing is the best  we can get dam n so be it. It is still 20x better than stock FW's... If I want better AM I can buy better or use my SDR's.

closed 2023-09-24 by OneOfEleven