OneOfEleven/uv-k5-firmware-custom #41: Rx expander destroys audio when FM

pawol, 2023-09-24:
Rx expander destroys audio when FM. Just enable RX or TX/RX in menu 31 and listen to FM station.

OneOfEleven, 2023-09-24:
I can't find an fm stations to listen too at the moment where I live.

But yes it will sound bad, the TX station needs to enable TX TX/RX mode it there end. They need to match to use the compander.

pawol, 2023-09-24:
Modulation should not impact compander. There should not be any difference between AM or FM because clear audio is processed independly in used modulation.

Raul-2017, 2023-09-27:
Hi. Yesterday I enabled TX+RX Compander in my two UV-K5s running your last firmware release.  I did a quick test talking each other on FM  2 meter band, and the result was the same as if no compander was set in TX:  received audio was super weak, hardly audible, as if volume had set to 0.  Then I just set compander to none, and voilá, perfect communication.  I think something could is wrong with this compander...

pawol, 2023-09-27:
Yes, audio on FM is completly unusable when compander is used. On AM works in RX mode. I did not checked audio quality when TX or TX/RX is used on AM

bricky149, 2023-09-30:
I'm under the impression the compander doesn't apply to AM at all, only FM, as I hear no difference between options when listening to a local airport. I confirm I hear no audio on my modded K5(8) when transmitting from my stock 5R+ with RX compander enabled, regardless of what ratio I choose in code.

I don't think it's a software bug though, perhaps a hardware limitation.

pawol, 2023-09-30:
Compander works in audio , not on modulated RF. So has to work in the same way on AM and FM.

bricky149, 2023-10-01:
The compander menu setting no longer appears for me, despite setting its flag to 1. I can't test any local changes.

closed 2023-10-03 by OneOfEleven