OneOfEleven/uv-k5-firmware-custom #43: why is sprinf can't process %f in this project

zhaoparrot, 2023-09-25:
firstly this is a 'me' problem,I feel sorry for border you, but I really can't figure out why. I been struggleing for severa hours.
in this project a code like this can't work:
	`sprintf(String, "%.5f", gSubMenuSelection * 1e-05);`
it will return a "f";
it have to be written like this :
`sprintf(String, "%d.%05u", gSubMenuSelection / 100000, abs(gSubMenuSelection) % 100000);`
do u know why?
it seems this project is using exactly the same printf with DualTachyon's project. It‘s just Not logical to have different result.
My apology for border u again.

OneOfEleven, 2023-09-25:
Floating point support was disabled before flash space was at it's limit. Simply enabling floating point support takes up a large chunk of flash.

closed 2023-09-25 by OneOfEleven

zhaoparrot, 2023-09-25:
Ah. I see now .thanks
in case other people wonder     this was controled in printf_config.h