OneOfEleven/uv-k5-firmware-custom #44: Latest build issue(OEFW-ed32177)

lohtse, 2023-09-25:
Ok so for the past 24 have compiled a few FW's with varying results... Some good some bad.. but the current build OEFW-ed32177  is by far the worst and oddest....  It wil work on AM for a few minutes before becoming very distorted to then NOT recieve anything again until I turn it off and on again... 

I have included a zip here so you can see if you have the issue...  I also included in the zip the last Good version that works frigging awesome for me too so you can compare and hopefully understand the issue I am referring too

[1of11 k5 Fws.zip](https://github.com/OneOfEleven/uv-k5-firmware-custom/files/12712175/1of11.k5.Fws.zip)

lohtse, 2023-09-25:
built same FW no FMradio and the issue stops!

OneOfEleven, 2023-09-25:
Yes it is doing something not right - the .bin you posted.

I'll test with the FM radio included, should be easy to find the problem now I know it happens when the FM radio is enabled.

OneOfEleven, 2023-09-25:
If I press the monitor button to disable the squelch in AM mode, the background hiss occasionally drops in level and then back up again as well, which is not right..

lohtse, 2023-09-25:
this is my default setup
[image]

OneOfEleven, 2023-09-25:
I see you're enabling most of the options, surprised it fits in the flash now.

Do have dual-watch enabled or channel scanning running when the problem happens ?

lohtse, 2023-09-25:
DW is enable(Always is) and no to channel scan, I found your FW doesn't quite like it,, Always heavy static for first second or 2 then clears but by then they stp[ trans and scan moves on lol

OneOfEleven, 2023-09-25:
When the problem happens is it when your listening to an FM radio station ?

lohtse, 2023-09-25:
> When the problem happens is it when your listening to an FM radio station ?

Airband...

OneOfEleven, 2023-09-25:
I don't suppose you can do a short video showing the problem ?

lohtse, 2023-09-25:
> I don't suppose you can do a short video showing the problem ?

May have to be tomorrow it thats ok... Currently on duty so may not have the time... But I will do them as soon as possible

OneOfEleven, 2023-09-25:
I can't duplicate the problem.

Do you have any kind of radio transmitters (broadcast or 2-way communications etc) in the vicinity (say 500 yards/meters) when the problem occurs ?

lohtse, 2023-09-25:
Hi Buddy

Apologies for late reply...

In answer to you question  I am a SAR team leader lol and use comms constantly so yes we do BUT the issue only started occuring AFTER the 24/09/23 v2 FW   so it is not an issue with other trans in the area etc..

OneOfEleven, 2023-09-25:
Just that if I can't reproduce the problem, the only way forward for me is to get as much detail as I can about what exactly is happening in order to try and find the cause.

lohtse, 2023-09-25:
> Just that if I can't reproduce the problem, the only way forward for me is to get as much detail as I can about what exactly is happening in order to try and find the cause. It's how the development/design process works.

I understand how the process works... which is why I made sure I discounted and local antennas.....   Been using the fw OEFW-4ea3792 on and off throught today and next too OEFW-b863143 it is one of the best I have used so far.. Both are virtually perfect. OEFW-b863143 doesn't hav ethe visual look and a little quiet but its great... I actually Wish I had taken/made a branch of both so I could try working with them for future use... Everything else has been either okish or not useable...   What ever you did in those two version was pure magic...

OneOfEleven, 2023-09-25:
Trying to find that version in the github commits to see what I changed.

Here's how it works for me, though no super strong signals at the mo to test with ..

https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/51590168/7c041da3-f45d-4414-8d3d-3df32ccbef32

lohtse, 2023-09-25:
> Trying to find that version in the github commits to see what I changed.
> 
> Here's how it works for me, though no super strong signals at the mo to test with, this is with the FMRadio enabled and running, the airband cuts in OK, am trying to break it but can't, I've edited it down and compressed it a lot though for github ..
> 
>  20230925_170727-1.mp4

am getting slightly lost here..

which firmware was thats?

OneOfEleven, 2023-09-25:
Now, the current latest

OneOfEleven, 2023-09-25:
I've also ran the firmware you posted, couldn't get that to break either. A bug that's hiding well !

OneOfEleven, 2023-09-25:
Not heard from anyone else about the AM fix for a while, so was thinking I've done it !

lohtse, 2023-09-25:
https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/48331155/851a86e2-d9c7-414d-986d-5e8cdae7df4e

this was taken at around 1530... was the latest at that time...  am probably around 40km from any antenna. other than our own repeaters.. the noise at the beginning of the trans actually gets longer and worse(thought the vid was longer sorry)

lohtse, 2023-09-25:
> Not heard from anyone else about the AM fix for a while, so was thinking I've done it !

TO be fair it generally works fine and like I said the version mentioned above I love it awesome. I just wish I could as some customisations etc too it now.

OneOfEleven, 2023-09-25:
The crackle/distortion at the start of their transmissions is when the code initially adjusts the RF gain, it has to find the correct gain setting to use with each new transmission, it takes a 100ms or so to do that, it's a side effect of having to botch what the RF chip won't properly do for itself (AGC) and the rate at which my code is called (once every 10ms).

OneOfEleven, 2023-09-25:
I'll try something, hold on

lohtse, 2023-09-25:
> The crackle/distortion at the start of their transmissions is when the code initially adjusts the RF gain, it has to find the correct gain setting to use with each new transmission, it takes a 100ms or so to do that, it's a side effect of having to bodge what the RF chip won't properly do for itself (AGC) and the rate at which my code is called (once every 10ms).
> 
> The QS firmware sounds like the start all the time.

trouble is it then continues to worsen now lol

any as to QS Fw   lets not talk about that... lol it doesn't even compare to todays versions of OEFW!

OneOfEleven, 2023-09-25:
I've uploaded a test version, try it when your back home or whenever. Let me know.

lohtse, 2023-09-25:
> I've uploaded a test version, try it when your back home or whenever. Let me know.

just flashed it... works and not as bad as last ones bugt I get more background hissing now

OneOfEleven, 2023-09-25:
Try using narrow mode for AM (W/N menu option) - it sets the RX bandwidth in both FM and AM modes.

lohtse, 2023-09-25:
> Try using narrow mode for AM (W/N menu option) - it sets the RX bandwidth in both FM and AM modes.

Thank you or the suggestion... actual makes no difference BUT what ever you are currently changing withing the FW  is making sight dirrences each time... Not quite as good as OEFW-4ea3792 yet but getting there again

lohtse, 2023-09-26:
fw OEFW-4ea3792

https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/48331155/7fdf03b0-5446-47f3-9601-8c24084b9df9

lohtse, 2023-09-26:
FW OEFW-b9ab214  (latest as of the comment)

https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/48331155/401f1972-7f63-4301-8ee5-9d7868f646de

lohtse, 2023-09-26:
Is there anyway you can revert back to setting that stopped the static/distortion before each trans??? this had been achieved as you can see from first video...

OneOfEleven, 2023-09-26:
I can revert but I prefer to better it/fix/improve.

Try the latest.

OneOfEleven, 2023-09-26:
Videos help a lot !   far better than words/descriptions. thank you.

lohtse, 2023-09-26:
Apologies for late response... Been a very busy day on callouts sadly...

Anyway just flashed the latest FW as of this comment  and damn its ver close the fw OEFW-4ea

https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/48331155/f2b49fcc-9285-42e4-b72e-5d51fdd12f63

3792

Still has a ot of exra noise and static but its by far the best since fw OEFW-4ea3792

OneOfEleven, 2023-09-26:
I can't decide where that rasping noise is coming from.
Disable the AM_fix option in the menu, then see if the noise is still there when the station comes up again (same place for radio etc).

OneOfEleven, 2023-09-26:
I discovered a strange compiler bug/problem early today which has been causing all sorts of problems with me getting the AM fix working.
I've never come across this compiler problem before so never occurred to me it could/was happening. Anyway, I've overcome it but the strange problem could cause all sorts of problems for anyone else writing C code :(

It's either a compiler problem or I'm missing something about C. Very strange.

lohtse, 2023-09-26:
> I can't decide where that rasping noise is coming from. Disable the AM_fix option in the menu, then see if the noise is still there when the station comes up again (same place for radio etc).

As instructed Am fix disabled

https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/48331155/e6a85807-af34-4d8d-a4c6-54e25c746f39

OneOfEleven, 2023-09-26:
oh, I forgot the AM demodulator would be saturated/clip. my bad.

Can you do it again but with a weaker signal, one that comes through OK on AM with the fix still disabled ?  .. maybe turn/rotate the radio around to weaken the signal ?

lohtse, 2023-09-26:
> oh, I forgot the AM demodulator would be saturated/clip. my bad.
> 
> Can you do it again but with a weaker signal, one that comes through OK on AM with the fix still disabled ? .. maybe turn/rotate the radio around to weaken the signal ?

I can do but will take long to take video as most round me are strng to medium ue to location... Also I am a little confused as too the need...  as the firmware thats  basically perfect has no issues with any Frqs.. So given recent builds now have issues I would have thought the Frg/strength is not the issue it is more an issue with a conflicting variable etc.

OneOfEleven, 2023-09-26:
I wanted to know if that noisy crackle was still there with the AM fix disabled, but need a weaker AM signal to hear whether it's still there or not. Nothing else wrong now I don't think. It works floorlessly here for me, don't know about for anyone else.

lohtse, 2023-09-26:
did you change anything in the very latest commits??? Just flashed and that seems a little better still noise but not as much.

OneOfEleven, 2023-09-26:
I slightly changed some of the tables in the AM fixer to get round a very strange but serious problem the compiler seems to have.
It made a huge difference to the AM fixer, I never knew about the compiler bug until today, it's partly why it's taken me so long to finish the AM fixer upper..

lohtse, 2023-09-26:
> I slightly changed some of the tables in the AM fixer to get round a very strange but serious problem the compiler seems to have. It made a huge difference to the AM fixer, I never knew about the compiler bug until today, it's partly why it's taken me so long to finish the AM fixer upper..

Well you are a damn wzard buddy... I hope it is for fulling for you to know you took a reasonable cheap HT and made it a good one ...

OneOfEleven, 2023-09-26:
I like coding, it's free :)

I'm actually an electronics hardware/software RF/digi/analog design engineer, I started with an apprenticeship at 15 when I left school in the 70's. I still love it now, but these days I do it for fun and for something to do.
Saying that, this firmware is only possible because of DualTachyon, they are the empowerer of all this :)

closed 2023-09-28 by OneOfEleven