OneOfEleven/uv-k5-firmware-custom #45: Dual Watch-related proposals

LZ2DMV, 2023-09-25:
Two things related to Dual Watch that might be good to have:

1) Option in the menu to adjust the Dual Watch interval (how fast the switching between the two monitored VFOs happen). It can be controlled from [here](https://github.com/OneOfEleven/uv-k5-firmware-custom/blob/ed321778c0b3fe6dfa5bc3585e30cc846260369e/app/app.c#L105) .
2) Virtual VFOs - split one VFO in two and use RSSI sampling to implement virtual Dual Watch between the two selected channels. Use menu option to pick the two channels part of the virtual VFO. Keep the regular dual watch between the two default VFOs.
That way, we can have 4 channels monitored in the same time with a minimal trade off for the switching time, because the virtual VFOs will be RSSI sampling-based instead of switching the entire VFO like in the regular scanner mode.

OneOfEleven, 2023-09-25:
There isn't really two VFO's, the radio only has one receiver which is scanned between the two 'virtual' VFO's. It's not really a dual-RX radio.

OneOfEleven, 2023-09-25:
I was thinking of adding some time settings in the menu, like how long to pause before resuming scanning after a signal vanishes.

I have to watch how much flash space I'm using up though, I want to add the band scope, but need to finish the AM fix first, seems it has a new bug I need to find.

LZ2DMV, 2023-09-25:
> There isn't really two VFO's, the radio only has one receiver which is scanned between the two 'virtual' VFO's. It's not really a dual-RX radio.

Yes, i am aware, but my idea is do further split the already virtual VFO into two more virtual VFOs, so a total of 4 channels to monitor at once in standard mode (2 VFOs x 2 virtual-virtual VFOs each). Of course, it is up to you to decide if it is worth it ans if it is useful at all.

But it will be useful to have a menu setting to tweak the interval on which Dual Watch probes the two VFOs (the place in the code i referenced above).

OneOfEleven, 2023-09-25:
I'll have a think about it, bare with me.

The dual-watch scan rate is set by "dual_watch_count_toggle_10ms", I've been moving all the time settings to the top of misc.c.

The toggle time is currently 100ms between the two vfo's.

LZ2DMV, 2023-09-25:
> The dual-watch scan rate is set by "dual_watch_count_toggle_10ms", I've been moving all the time settings to the top of misc.c.
> 
> The toggle time is currently 100ms between the two vfo's.

Awesome! Thank you for making it easily adjustable.

closed 2023-09-26 by LZ2DMV