OneOfEleven/uv-k5-firmware-custom #47: Audio in TX is too dark.

Braccato, 2023-09-25:
Please fix the audio in TX is too dark.

OneOfEleven, 2023-09-25:
Don't know what you mean by 'too dark' ?

There's a menu entry for adjusting the mic gain if that's what you mean ?

INDIAJULIETVICTOR, 2023-09-25:
dark modulation means that there is a predonimance of low frequencies that make modulation as if you were talking in a can

OneOfEleven, 2023-09-25:
I've not altered the TX audio filtering, does it sound 'dark' for anyone else ?

Braccato, 2023-09-25:
The audio in TX comes out too bad, please fix it only you can do it

INDIAJULIETVICTOR, 2023-09-25:
The strange thing is that with the firmware : Ver26_rssi78_bar_vol_sq_XSX_V3_5_2 
the modulation in Tx is excellent with all the other OEFW no. 
I tried to search the Beken registers which one could be responsible for this problem but I was unable to find it.
I know that you have the Magic Touch... :-)

OneOfEleven, 2023-09-25:
Try the bin I just uploaded, but looking at the TX audio it looks the same as QS's original firmware on the screen here.

OneOfEleven, 2023-09-25:
All the TX audio registers are as per original, I've never knowingly changed them, but you never know.

INDIAJULIETVICTOR, 2023-09-25:
The strange thing is that with firmware stocks this problem is not highlighted, maybe there is some register dedicated to TX modulation to be set ? HPF, DSP ...??
I tried with register 43 (AFTxLPF2 filter Band Width) but I did not get any results, I also tried to increase the bandwidth, but also here nothing to do.

OneOfEleven, 2023-09-25:
Does DualTachyon's firmware sound OK ?

OneOfEleven, 2023-09-25:
I'm just going back through the github uploads, I'll find where the tx audio was altered and see  what i change.

I'll find it don't worry.

pawol, 2023-09-25:
@maybe  expandor routines impacts mic

OneOfEleven, 2023-09-25:
No, it's something I would never of thought of. Enabling the chips DTMF decoder affects the TX audio bandwidth. It shouldn't but it does.

Anyway, I've fixed it now.

The compander is fine to enable/use.

OneOfEleven, 2023-09-25:
Not having a proper datasheet for the BK4819 chip is a bum pain.

INDIAJULIETVICTOR, 2023-09-25:
I have tried the latest version of DualTachyon just now, it is OK good modulation.

OneOfEleven, 2023-09-25:
Github has it's uses !

closed 2023-09-25 by OneOfEleven

INDIAJULIETVICTOR, 2023-09-25:
I knew you were magical... :-D it is perfect now !!! 

THANK YOU VERY MUCH !!!!

OneOfEleven, 2023-09-25:
I'm not lol, I'm just methodical/determined, and like to help the muggles ;)

egzumer, 2023-09-25:
To be honest I don't know which one is better. If that was discovered, maybe something in the middle would be optimal if that is possible. People were complaining about audio modulation quality of this radio.

Below samples were recorder with SDRSharp and RTL-SDR:
[audio test.zip](https://github.com/OneOfEleven/uv-k5-firmware-custom/files/12719857/audio.test.zip)

OneOfEleven, 2023-09-25:
The first bassier one corrupts the CTCSS/DCS part of the spectrum (270Hz and below), so would harm the radio links that use CTCSS (analogue sub audio tone squelch) and/or DCS (digital squelch).

The 2nd one doesn't corrupt CTCSS/DCS, and has better/punchier intelligibility at the receiving end.

OneOfEleven, 2023-09-25:
[image]

egzumer, 2023-09-25:
Well I'm no ham operator, I don't know how a good radio sounds on the other end. I get the CTCSS part, and intelligibility, you are right about that, but if the TX audio bandwidth is something that can be adjusted, maybe it is something worth noting for the future. Just wondering.

OneOfEleven, 2023-09-25:
It's quite adjustable in various ways, but you have to stay within normal specifications for transmission such as limiting TX bandwidth (otherwise you can splatter across adjacent RF channels).
But yes the TX and RX audio filters, TX deviation and RX bandwidth can all be adjusted.