OneOfEleven/uv-k5-firmware-custom #50: Strange compiler problem

OneOfEleven, 2023-09-26:
I've been coding in C for a long long time now, but I've recently come across a problem with two very different C/C++ compilers.

I have some small const tables in my code ..

```
static const int16_t lna_short_dB[] = {-33, -30, -24, 0};
static const int16_t lna_dB[]       = {-24, -19, -14,  -9, -6, -4, -2, 0};
static const int16_t mixer_dB[]     = {-8, -6, -3, 0};
static const int16_t pga_dB[]       = {-33, -27, -21, -15, -9, -6, -3, 0};
```

If you single step when your code reads those values, they are not always the correct values.

If I surround the negative values (all ints) with brackets like this ..

```
static const int16_t lna_short_dB[] = { (-33), (-30), (-24),  0};
static const int16_t lna_dB[]       = { (-24), (-19), (-14), ( -9), (-6), (-4), (-2), 0};
static const int16_t mixer_dB[]     = { ( -8), ( -6), ( -3),    0};
static const int16_t pga_dB[]       = { (-33), (-27), (-21), (-15), (-9), (-6), (-3), 0};
```

the code reads the values from the tables correctly.

Am I missing something ??

INDIAJULIETVICTOR, 2023-09-26:
never seen before

bricky149, 2023-09-30:
Had a look myself and can find no difference in GCC's IR between using brackets or not. The md5sum is the same between binaries too.

OneOfEleven, 2023-09-30:
Okey doke, interesting. Maybe because I'm not using the latest compiler version, am at least two major versions behind current compiler release.

zhaoparrot, 2023-10-12:
> Current latest compile release creates larger .bin files (they didn't originally fit into the CPU), so had to use an old compiler version.

Ah，that may be why my bin file is so big. I wonder how much difference it make.

OneOfEleven, 2023-10-14:
At the the time DualTachyon had mostly complete his firmware decompile, the compiled size was right at the limit of what flash is available, I think it was only like 1kB difference or so compile size between the compiler versions, but that was enough to make all the difference (at the time).

closed 2023-10-24 by OneOfEleven