OneOfEleven/uv-k5-firmware-custom #51: cb92f2e

M7OCM, 2023-09-26:
Hi 111, good work with the AM tweaks current works good for me on civil and mil airband.

Menu issue to report. Sporadic.

eg Start at SQL (after radio boot). Then press 12 WIDE,  then 18 CH-DIS etc etc eventually pressing any combo of numbers fails to switch to the menu item. Tried a few times different combos and it happens sooner than other times (average seems to be 5 menu selections one after the other). Rebooting radio each test.

M7OCM, 2023-09-26:
Will do just waiting on the file to upload then download lol

M7OCM, 2023-09-26:
https://youtube.com/shorts/OyBN9uQpWBA?si=1uSJ3oMmzoYGguFQ
Change of plan stuck it up on YT quicker solution

First few presses ok, then trying to enter 14 etc not working.

OneOfEleven, 2023-09-26:
ah !

The menu has 53 items, you tried to enter a menu number above 53, you pressed 8 then tried to enter the 2nd number, but it won't accept menu 80, 81, 82 etc. So if you start a menu number with 6, 7, 8, or 9 it won't work.

I can fix that. Bare with me.

OneOfEleven, 2023-09-26:
I've upload fixed firmware. Should be good now, you can just keep entering any numbers you like.

M7OCM, 2023-09-26:
> I've upload fixed firmware. Should be good now, you can just keep entering any numbers you like.

Nice fix thank you, off to punch some more random squeaky menu numbers lol

OneOfEleven, 2023-09-26:
lol, yes the buttons have gone squeaky on my radio too.

closed 2023-09-26 by OneOfEleven