OneOfEleven/uv-k5-firmware-custom #52: radio FM menu

Yag00000, 2023-09-26:
add menu radio Fm+0

pawol, 2023-09-26:
FM radio is disabled on compilation stage. Yo can compile FM radio enabled oneself. By the way, it could be enabled by default because very low used flash space.

OneOfEleven, 2023-09-26:
> FM radio is disabled on compilation stage to save space. You can compile FM radio enabled oneself. By the way, it could be enabled by default because very low used flash space.

Enabling the FMRADIO takes quite a lot - 4k flash bytes, that's a lot when we're already near the limit.

lohtse, 2023-09-26:
> > FM radio is disabled on compilation stage to save space. You can compile FM radio enabled oneself. By the way, it could be enabled by default because very low used flash space.
> 
> Enabling the FMRADIO takes quite a lot - 4k flash bytes, that's a lot when we're already near the limit.

already dangerously close to the limits for me

text    data     bss     dec     hex filename
  60004       0    2756   62760    f528 firmware

OneOfEleven, 2023-09-26:
oh right, I'll have to see what I can space optimize then.

closed 2023-09-27 by OneOfEleven