OneOfEleven/uv-k5-firmware-custom #53: Force narrow bandwidth when AM

pawol, 2023-09-26:
Force narrow bandwidth when AM because AM is able to pass only narrow audio.  Wide bandwidth increases hissing only on AM

OneOfEleven, 2023-09-26:
Yes you're right.

The reason AM is still used for aircraft is because it allows you to hear any other transmitting aircraft weaker than the main signal, they can't tell what the weaker signal are saying but the tower hears the heterodyne tone, so AM allows the tower to know if a weaker signal is trying to get through (might be an emergency).
FM doesn't work like that, FM has a threshold thing which masks/hides weaker signals.

They were also taking advantage of the heterodyne effect by having aircraft offset their transmit frequencies by a kHz or so in order to deliberately create the heterodyne tone.
I don't know if they still do that though, but it means wide band mode is needed if they do still do it (to capture the offset transmissions as well as the on-frequency transmissions).

But yes, I do set the airband channels to narrow mode myself too.

OneOfEleven, 2023-09-26:
I don't want to force narrow mode though, I'm all for giving the user as many choices/options as possible.

M7OCM, 2023-09-27:
> I don't want to force narrow mode though, I'm all for giving the user as many choices/options as possible.

Yes, if it were a user option during compilation maybe ,but definitely AM needs WAM for other areas such as milair which works much better than NAM. Incidentally, owning quite a few radios a lot RX airband better in WAM. Yes heterodyne is still used very effectively in ATC.

lohtse, 2023-09-27:
> Force narrow bandwidth when AM because AM is able to pass only narrow audio. Wide bandwidth increases hissing only on AM

It should be a user option not forced. Depending n location wide can work better than narrow... It could be OneOfELeven as an enable flag in mage file but I see little point to it as will many... After all you can just set it in the menus and in CPS/Chirp yourself anyway.

pawol, 2023-09-27:
Worth to make bandwidth even narrow as 6,25kHz for AM. Wideband is against theory for AM

OneOfEleven, 2023-09-27:
It's possible, will have to see where to save the extra config bit needed.

closed 2023-09-29 by OneOfEleven