OneOfEleven/uv-k5-firmware-custom #56: compile failed with FMRADIO enabled

troilus, 2023-09-27:
compile failed with FMRADIO enabled
···
GIT_HASH = b0794ec
arm-none-eabi-gcc -Os -Wall -Werror -mcpu=cortex-m0 -fno-builtin -fshort-enums -fno-delete-null-pointer-checks -std=c11 -MMD -DPRINTF_INCLUDE_CONFIG_H -DGIT_HASH=\"b0794ec\" -DENABLE_FMRADIO -DENABLE_OVERLAY -DENABLE_UART -DENABLE_SMALL_BOLD -DENABLE_VOICE -DENABLE_ALARM -DKEEP_MEM_NAME -DENABLE_WIDE_RX -DENABLE_CTCSS_TAIL_PHASE_SHIFT -DENABLE_MAIN_KEY_HOLD -DENABLE_BOOT_BEEPS -DENABLE_COMPANDER -DENABLE_SHOW_CHARGE_LEVEL -DENABLE_REVERSE_BAT_SYMBOL -DENABLE_NO_SCAN_TIMEOUT -DENABLE_AM_FIX -DENABLE_AM_FIX_SHOW_DATA -DENABLE_AUDIO_BAR -DENABLE_COPY_CHAN_TO_VFO -I /home/uv-k5-firmware-custom -I /home/uv-k5-firmware-custom/external/CMSIS_5/CMSIS/Core/Include/ -I /home/uv-k5-firmware-custom/external/CMSIS_5/Device/ARM/ARMCM0/Include -c version.c -o version.o
arm-none-eabi-gcc -mcpu=cortex-m0 -nostartfiles -Wl,-T,firmware.ld start.o init.o sram-overlay.o external/printf/printf.o driver/adc.o driver/aes.o driver/backlight.o driver/bk1080.o driver/bk4819.o driver/crc.o driver/eeprom.o driver/flash.o driver/gpio.o driver/i2c.o driver/keyboard.o driver/spi.o driver/st7565.o driver/system.o driver/systick.o driver/uart.o app/action.o app/app.o app/dtmf.o app/fm.o app/generic.o app/main.o app/menu.o app/scanner.o app/uart.o am_fix.o audio.o bitmaps.o board.o dcs.o font.o frequencies.o functions.o helper/battery.o helper/boot.o misc.o radio.o scheduler.o settings.o ui/battery.o ui/fmradio.o ui/helper.o ui/inputbox.o ui/lock.o ui/main.o ui/menu.o ui/rssi.o ui/scanner.o ui/status.o ui/ui.o ui/welcome.o version.o main.o -o firmware
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: app/fm.o: in function `FM_Tune':
fm.c:(.text+0x190): undefined reference to `gScheduleFM'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: app/fm.o: in function `FM_Key_UP_DOWN':
fm.c:(.text+0x284): undefined reference to `gRequestSaveFM'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: app/fm.o: in function `FM_PlayAndUpdate':
fm.c:(.text+0x2b0): undefined reference to `SETTINGS_SaveFM'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: fm.c:(.text+0x2d8): undefined reference to `gScheduleFM'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: app/fm.o: in function `FM_ProcessKeys':
fm.c:(.text+0x5f6): undefined reference to `ACTION_FM'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: fm.c:(.text+0x67c): undefined reference to `gAnotherVoiceID'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: fm.c:(.text+0x684): undefined reference to `gRequestSaveFM'
collect2: error: ld returned 1 exit status
make: *** [Makefile:272: firmware] Error 1
···

lohtse, 2023-09-27:
I am 100% certain it is not the fact FMradio is enabled as EVERY single one of my OneOfElevens  FWs has FMradio enabled and has no issues ever.

OneOfEleven, 2023-09-27:
Can you copy/paste the list of enable items from the top of your makefile ?

lohtse, 2023-09-27:
[image]
 as you can see I have most this enable without issue

> custom/external/CMSIS_5/Device/ARM/ARMCM0/Include -c version.c -o version.o

have even added the bin file as proof

lohtse, 2023-09-27:
[firmware.packed.zip](https://github.com/OneOfEleven/uv-k5-firmware-custom/files/12739947/firmware.packed.zip)

troilus, 2023-09-27:
This is my Makefile, when ENABLE_FMRADIO = 0 and compile successful

```
# compile options (see README.md for descriptions)
# 0 = disable
# 1 = enable
#
ENABLE_SWD                    := 0
ENABLE_OVERLAY                := 1
ENABLE_UART                   := 1
ENABLE_AIRCOPY                := 0
ENABLE_FMRADIO                := 1
ENABLE_NOAA                   := 0
ENABLE_VOICE                  := 1
ENABLE_ALARM                  := 1
ENABLE_BIG_FREQ               := 0
ENABLE_SMALL_BOLD             := 1
ENABLE_KEEP_MEM_NAME          := 1
ENABLE_WIDE_RX                := 1
ENABLE_TX_WHEN_AM             := 0
ENABLE_CTCSS_TAIL_PHASE_SHIFT := 1
ENABLE_MAIN_KEY_HOLD          := 1
ENABLE_BOOT_BEEPS             := 1
ENABLE_COMPANDER              := 1
ENABLE_SHOW_CHARGE_LEVEL      := 1
ENABLE_REVERSE_BAT_SYMBOL     := 1
ENABLE_NO_SCAN_TIMEOUT        := 1
ENABLE_AM_FIX                 := 1
ENABLE_AM_FIX_SHOW_DATA       := 1
ENABLE_SQUELCH1_LOWER         := 0
ENABLE_AUDIO_BAR              := 1
ENABLE_COPY_CHAN_TO_VFO       := 1
#ENABLE_SINGLE_VFO_CHAN       := 1
#ENABLE_BAND_SCOPE            := 1
```

troilus, 2023-09-27:
git pull                                                                                                                                                       [22:47:22]
Already up to date.

OneOfEleven, 2023-09-27:
hhmmm, another puzzle to solve

troilus, 2023-09-27:
Strange, Whan else can I offer?

troilus, 2023-09-27:
> > This is my Makefile, when ENABLE_FMRADIO = 0 and compile successful这是我的生成文件，当 ENABLE_FMRADIO = 0 并且编译成功时
> > ```
> > # compile options (see README.md for descriptions)
> > # 0 = disable
> > # 1 = enable
> > #
> > ENABLE_SWD                    := 0
> > ENABLE_OVERLAY                := 1
> > ENABLE_UART                   := 1
> > ENABLE_AIRCOPY                := 0
> > ENABLE_FMRADIO                := 1
> > ENABLE_NOAA                   := 0
> > ENABLE_VOICE                  := 1
> > ENABLE_ALARM                  := 1
> > ENABLE_BIG_FREQ               := 0
> > ENABLE_SMALL_BOLD             := 1
> > ENABLE_KEEP_MEM_NAME          := 1
> > ENABLE_WIDE_RX                := 1
> > ENABLE_TX_WHEN_AM             := 0
> > ENABLE_CTCSS_TAIL_PHASE_SHIFT := 1
> > ENABLE_MAIN_KEY_HOLD          := 1
> > ENABLE_BOOT_BEEPS             := 1
> > ENABLE_COMPANDER              := 1
> > ENABLE_SHOW_CHARGE_LEVEL      := 1
> > ENABLE_REVERSE_BAT_SYMBOL     := 1
> > ENABLE_NO_SCAN_TIMEOUT        := 1
> > ENABLE_AM_FIX                 := 1
> > ENABLE_AM_FIX_SHOW_DATA       := 1
> > ENABLE_SQUELCH1_LOWER         := 0
> > ENABLE_AUDIO_BAR              := 1
> > ENABLE_COPY_CHAN_TO_VFO       := 1
> > #ENABLE_SINGLE_VFO_CHAN       := 1
> > #ENABLE_BAND_SCOPE            := 1
> > ```
> 
> Other than beeps and Alarm enabled you are the same as me... And as stated I Always enable FMradio... never once had an issue除了哔哔声和启用警报之外，您和我一样...如前所述，我总是启用FMradio...从来没有遇到过问题

Could you please upload the bin compiled with my config, Thanks.

OneOfEleven, 2023-09-27:
I would say, try downloading the zip file directly from github (not using git) and see if that compiles ok. Part from that, it's a puzzle to me at the moment.

egzumer, 2023-09-27:
Call `make clean`
Most probably you have old object files that didn't get rebuilt.

closed 2023-09-27 by troilus