OneOfEleven/uv-k5-firmware-custom #57: strange behavior of the external PTT

INDIAJULIETVICTOR, 2023-09-27:
the external PTT stops working if the channel opens squelch, while with the inserted squelch it works correctly, please look the video.

https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/145300322/aa6f0622-e47b-49cd-a7d5-bfa345464174

OneOfEleven, 2023-09-27:
Another puzzle !

I've tried the original QS, it's the same.

Seems like BCLO (Busy Channel Lock Out) is enabled for an external mike. hmmm

OneOfEleven, 2023-09-27:
Does the USB charge indicator on the top status right by the battery level bar show up when you press the external PTT with the squelch opened (monitor enable) ?

INDIAJULIETVICTOR, 2023-09-27:
No, but the blue led turn on..

https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/145300322/88a182e8-98e5-4968-bb00-8a70f8fd927b

INDIAJULIETVICTOR, 2023-09-27:
also with SQL 0 the same problem.

OneOfEleven, 2023-09-27:
Your USB charger indicator shows up too.

Not sure if I can solve this one, time will tell. As the original QS firmware does the same too it's going to be a problem.

OneOfEleven, 2023-09-27:
If I turn the volume right up with an external mic plugged and squelch open, the PTT works - strange.

I'm trying a baofeng mic with the radio. I'm wondering if they are slightly different to a Quensheng external mic.

pawol, 2023-09-29:
On the original qQansheng ext mic there is no problem

pawol, 2023-09-29:
I got ext mic bundled with k5

OneOfEleven, 2023-09-29:
A schematic of the microphone would be useful !

closed 2023-09-29 by OneOfEleven

Lar-Sen, 2023-10-16:
I am concerned with the same bug using factory firmware. I guess this has something to do with key debouncing function, but I wasn't able to track the bug down in IDA/source code. Different behaviours should be explained by the fact that some headsets may have better circuitry (capacitors) to prevent excessive bouncing.

May I have a hint on own this issue has been resolved?  Thanks for your (hard) work by the way :)

Lar-Sen, 2023-10-16:
Thank you for your quick answer. I thought the problem was fixed in your fork.
My own headset come from an old unit I have, a Wouxun KG-UVD1P, though I suppose it is not genuine anyway. The kit came from Radioshop888 (in 2009!).
I will try to write a schematic if I manage to open its case without breaking everything.