OneOfEleven/uv-k5-firmware-custom #58: Is it possible to have rssi bar display function?

troilus, 2023-09-27:
Thanks for your custom firmware, is it possible to have rssi bar display? It would be useful.

Braccato, 2023-09-27:
I would also like the rssi bar like pietr022's v78

OneOfEleven, 2023-09-27:
Yes can do. Bare with me though, don't close this issue else I'll forget it's here.

zhaoparrot, 2023-09-28:
I also had this idea a few days ago , wanted to add RSSI_SBAR to the code, but after I looked into[ the code of piotr022](https://github.com/piotr022/UV_K5_playground/blob/main/src/rssi_sbar_hot/main.cpp), I didn't even know where to start.
I successfully integrated  fagci's spectrogram into my fork though, which occupied a large area of flash and prevented me from opening any other mods. Now even if I write two more lines of code, flash overflow alarm will appear😆.in order to revert the RX display pixel to make it more obvious，I have to delete part of fagci's function.

troilus, 2023-09-28:
May this help? @zhaoparrot 
https://github.com/OneOfEleven/uv-k5-firmware-custom/pull/1/files

zhaoparrot, 2023-09-28:
@troilus thanks for the link
I'm no expert on this 
I think it should be sth like this
```
const int16_t dBm = (gCurrentRSSI[gEeprom.RX_CHANNEL] / 2) - 160;
		if (dBm >= -83) {
			sprintf(String, "%-3d S9+ %d", dBm, gEeprom.RX_CHANNEL);
		}
		else if (dBm >= -93) {
			sprintf(String, "%-3d S9 %d", dBm, gEeprom.RX_CHANNEL);
		}
		else if (dBm >= -99) {
			sprintf(String, "%-3d S8 %d", dBm, gEeprom.RX_CHANNEL);
		}
		else if (dBm >= -105) {
			sprintf(String, "%-3d S7 %d", dBm, gEeprom.RX_CHANNEL);
		}
		else if (dBm >= -111) {
			sprintf(String, "%-3d S6 %d", dBm, gEeprom.RX_CHANNEL);
		}
		else if (dBm >= -117) {
			sprintf(String, "%-3d S5 %d", dBm, gEeprom.RX_CHANNEL);
		}
		else if (dBm >= -123) {
			sprintf(String, "%-3d S4 %d", dBm, gEeprom.RX_CHANNEL);
		}
		else if (dBm >= -129) {
			sprintf(String, "%-3d S3 %d", dBm, gEeprom.RX_CHANNEL);
		}
		else if (dBm >= -135) {
			sprintf(String, "%-3d S2 %d", dBm, gEeprom.RX_CHANNEL);
		}
		else if (dBm >= -141) {
			sprintf(String, "%-3d S1 %d", dBm, gEeprom.RX_CHANNEL);
		}
		else {
			sprintf(String, "%-3d S0 %d", dBm, gEeprom.RX_CHANNEL);
		}
		UI_PrintStringSmall(String, 2, 0, 3);
```
but it's not working right 
after enter RX mode  it stops update
you can put this code in UI/main.c
```
	if (center_line_is_free)
	{	// we're free to use the middle empty line for something
```
for testing 
let me know if you make any progress
have fun

closed 2023-09-29 by OneOfEleven

zhaoparrot, 2023-09-29:
when receiving from nearby device dBm may have a postiove value and the bar will display empty,is this a bug ?

zhaoparrot, 2023-09-29:
I got it , it's a type error.  see [this pull request](https://github.com/OneOfEleven/uv-k5-firmware-custom/pull/70)

OneOfEleven, 2023-09-29:
I got my types a little mixed up, was a bit tired at the time :)