OneOfEleven/uv-k5-firmware-custom #59: Diminuiu a sensibilidade de RX.

marciolimapix, 2023-09-27:
Após instalação dessa firmware, a sensibilidade da recepção diminuiu, principalmente próximo aos 260MHz.

pawol, 2023-09-28:
To be honest I am in doubt that someone will be eager to translate your laguage if you do not want even try to write in english.

Para ser sincero, tenho dúvidas de que alguém esteja ansioso para traduzir seu idioma se você não quiser nem tentar escrever em inglês.

OneOfEleven, 2023-09-28:
I was going to answer but I wanted to check the radios specifications first. I think 260MHz is way way out of specification, in which the radio WILL be insensitive.

OneOfEleven, 2023-09-28:
oh, ignore that, it is suppose to be OK there. Anyway, I have other bugs to sort out first. A bit overloaded with stuff at the moment.

closed 2023-09-28 by marciolimapix