OneOfEleven/uv-k5-firmware-custom #60: LED Customisation suggestion

lohtse, 2023-09-27:
Is it Possible y=to change the LED light and function

EG Instead of Red for transmission it could be the Blue??

Just curious

OneOfEleven, 2023-09-27:
I'm currently trying to find how the blue LED is turned on, but no luck so far.

I can see the red and green pin details but not the blue.

LZ2DMV, 2023-09-27:
I think the blue LED cannot be controlled from software with the GPIO system. It is connected to the charging circuit on the hardware level.

OneOfEleven, 2023-09-27:
Yes seems that way.

lohtse, 2023-09-27:
> I think the blue LED cannot be controlled from software with the GPIO system. It is connected to the charging circuit on the hardware level.

I would disagree somewhat as it comes one when you connect the programming cable as well

LZ2DMV, 2023-09-27:
> > I think the blue LED cannot be controlled from software with the GPIO system. It is connected to the charging circuit on the hardware level.
> 
> I would disagree somewhat as it comes one when you connect the programming cable as well

The CH340 in the cable supplies 5 V that is fed into the charging circuit at some point, that's why. Poor design by Quansheng.

lohtse, 2023-09-27:
> > > I think the blue LED cannot be controlled from software with the GPIO system. It is connected to the charging circuit on the hardware level.
> > 
> > 
> > I would disagree somewhat as it comes one when you connect the programming cable as well
> 
> The CH340 in the cable supplies 5 V that are fed into the charging circuit at some point, that's why. Poor design by Quansheng.

OK I will accept what you are saying.but still hope for a possible solution

wrcrooks, 2023-09-27:
The wiki has where the LEDs are connected to.

https://github.com/ludwich66/Quansheng_UV-K5_Wiki/raw/main/hardware/Block_Diagram/Quansheng_UV-K5_Block_Diagram_v01.jpg

OneOfEleven, 2023-09-27:
Thank you for the block diagram.

The blue LED comes on and also a non-existent USB charger is detected when an external PTT is pressed when the squelch is open :(

OneOfEleven, 2023-09-27:
Really need a full schematic to find that one I think.

closed 2023-09-29 by OneOfEleven