OneOfEleven/uv-k5-firmware-custom #61: Disallow entering frequencies 630-840MHz

pawol, 2023-09-27:
BK chipsed has some gap in supported frequencies. If I remember it is 630-840MHz. It should not be possible to enter that range of frequencies into VFO

lohtse, 2023-09-27:
that is correct

OneOfEleven, 2023-09-27:
Seems to do be 630MHz to 757MHz'ish

I'll put a block on that range so the user can't set it there.

OneOfEleven, 2023-09-27:
The datasheet says 18~660 and 840~1300, but it stops at 630MHz, restarts around ~757MHz, very insensitive but then that's to be expected.

wrcrooks, 2023-09-27:
"World wide band: 18 MHz ~ 660 MHz, 840 MHz ~1300 MHz" - Datasheet

https://prkele.prk.tky.fi/~ftg/files/datasheets/BK4819_Datasheet_V1.01.pdf

OneOfEleven, 2023-09-27:
can't enter above 999.999 anyway

wrcrooks, 2023-09-27:
> can't enter above 999.999 anyway

Is that a UI limitation?

OneOfEleven, 2023-09-27:
yes

lohtse, 2023-09-27:
> can't enter above 999.999 anyway

yes you can, you have to use up and down to enter the frq after 999.9999

OneOfEleven, 2023-09-27:
oh yeah lol

never knew that !

OneOfEleven, 2023-09-27:
take it up to 1300. let me know what happens ;)

lohtse, 2023-09-27:
> oh yeah lol
> 
> never knew that !

Even I am still learning lol

wrcrooks, 2023-09-27:
Can we shrink the frequency text on the UI and increase the variable size to allow for >1GHz input?

lohtse, 2023-09-27:
> Can we shrink the frequency text on the UI and increase the variable size to allow for >1GHz input?

Something I was thinking too... Differnt text sizes(like Fagci's)  and even different fonts

OneOfEleven, 2023-09-27:
I'll let you design the slightly smaller font ;)  .. I did the small font but your turn for the bigger font

OneOfEleven, 2023-09-27:
fonts take up a lot of flash bytes

lohtse, 2023-09-27:
> I'll let you design the slightly smaller font ;) .. I did the small font but your turn for the bigger font

I prefer smaller text...makes tthings seem less crowded etc

lohtse, 2023-09-27:
you ever notice that your FW is never rebooted by k5prog_wn after flashing?

wrcrooks, 2023-09-27:
I haven't reviewed but there might not be any sub-1GHz checks in the code, it might just be that it places the first digit entered using the keypad in the _00.0000 place (100s of MHz)

OneOfEleven, 2023-09-27:
> you ever notice that your FW is never rebooted by k5prog_wn after flashing?

yes. I keep meaning to look into that

wrcrooks, 2023-09-27:
> you ever notice that your FW is never rebooted by k5prog_wn after flashing?

It reboots when I use the linux executable of k5prog

OneOfEleven, 2023-09-27:
I've been distracted by getting the AM fix 100% for past how ever days, and fixing the odd bug

OneOfEleven, 2023-09-27:
IT checks now, it will no longer allow less than 18MHz or between 630 to 757, I need to stop the user using the up/down buttons to bypass direct freq entry

OneOfEleven, 2023-09-27:
OK, I've uploaded new firmware that stops user going into the 630 to 757MHz range.

OneOfEleven, 2023-09-27:
> you ever notice that your FW is never rebooted by k5prog_wn after flashing?

I think I got it now. It should reboot the radio OK - let me know if it don't

pawol, 2023-09-28:
Now when I put 650.000MHz into VFO then frequency is changed onto 630.000MHz. If I know 630 is forbidden also. When I use arrows then maximum frequency is 629,99375 (below 630 as we see)

lohtse, 2023-09-28:
> > you ever notice that your FW is never rebooted by k5prog_wn after flashing?
> 
> I think I got it now. It should reboot the radio OK - let me know if it don't

It does...

OneOfEleven, 2023-09-28:
> Now when I put 650.000MHz into VFO then frequency is changed onto 630.000MHz. It is almost right behaviour. If I know 630 is forbidden also on BK. When I use arrows then maximum frequency is 629,99375 (below 630 as we see).
> 
> I see also that upper limit is set on 760MHz. Is it right? Wer talked about 757MHz at the beginning

Yes I was lazy last night and decided not to check the current frequency step value to choose the correct frequency to limit/actually use (instead of 630 or 629,99375).

I'll do that today, was a busy day yesterday.

OneOfEleven, 2023-09-28:
But yes, 630 is the chip won't do, so I need to just subtract the current frequency step value from 630 and limit it to there. I also chose 760MHz rather than 757, just to round it up a bit, Makes no odds really cuz the radio is as deaf as a stone there.

OneOfEleven, 2023-09-28:
OK, think I got it right now. Just uploaded.

pawol, 2023-09-28:
Upper limit is 757 or 760 ?

OneOfEleven, 2023-09-28:
I've gone for 840MHz as per datasheet, but I found the PLL starts to lock around 757MHz'ish

pawol, 2023-09-28:
Ticket can be closed because the fix works as expected

closed 2023-09-28 by OneOfEleven