OneOfEleven/uv-k5-firmware-custom #65: does not work  ENABLE_TX_WHEN_AM             := 0       allow TX (always FM) when RX is set to AM

lohtse, 2023-09-28:
Si as the title says

ENABLE_TX_WHEN_AM             := 0       allow TX (always FM) when RX is set to AM

Does not work

It will work until you recieve and AM TX after that it will continue to say TX disabled no matter what FRQ you are using. only way to resolve is to reboot radio

lohtse, 2023-09-28:
> if it's set to '0', then it should always say TX DISABLED if you try and transmit on a channel/freq that is set to AM mode ?

TX on an FM band.. will work if no AM trans or no AM trans for quite some time.. BUt if there has been  an AM tran it will either take 20-30s before I can TX or will not let me till I reboot radio... Noticed it the other day but forgot to mention. To day was using K5 and was listeninig on AM  and FM  but could not trans on FM band when I needed

OneOfEleven, 2023-09-28:
oh right, it wasn't letting you TX on an FM chan/freq .. okey doke

lohtse, 2023-09-28:
> oh right, it wasn't letting you TX on an FM chan/freq .. okey doke

It used to work.. so it is a recent thing(I think lol)

OneOfEleven, 2023-09-28:
I did recently edit that bit of code, I'll go look in a bit, doing too many things at once at the mo

lohtse, 2023-09-28:
> I did recently edit that bit of code, I'll go look in a bit, doing too many things at once at the mo

LOL thats what happens when you have possible the best FW for the K5/6 range lol

Whilst it is import, Just take your time buddy..  I can work with it for now...

OneOfEleven, 2023-09-28:
:)

I think I've fixed it, let me know if not

lohtse, 2023-09-28:
> :)
> 
> I think I've fixed it, let me know if not

stows as TXing but nothing is actually happening

OneOfEleven, 2023-09-28:
I don't suppose you can do a quick video ?

lohtse, 2023-09-28:
> I don't suppose you can do a quick video ?

just flashing stock.27

lohtse, 2023-09-28:
confirmed the issue is still on .27  I suspect it possiblydue to switching modulation from FM to AM and back.. so possible not fixable

lohtse, 2023-09-28:
slightly offtopic but just been posted in telegram grou

Bricky, [28/09/2023 19:06]
Think I noticed a mistake in 1o11's lower squelch-1 code. The close RSSI is bigger than the open when it's not airband. Changing it to be smaller makes my radio closer to what my stock 5R picks up.

Bricky, [28/09/2023 19:07]
I'm glad I have a supermarket warehouse nearby to test it out. Spent way too much time trying to make it better.

OneOfEleven, 2023-09-28:
The NOISE and GLITCH squelch setting are the inverse of the RSSI squelch.
so ..
 Lower RSSI value = more sensitive squelch.
 Higher NOISE value = more sensitive squelch.
 Higher GLITCH value = more sensitive squelch.

bricky149, 2023-09-30:
Not sure if I'm missing something but I can't reproduce this on latest git build.

lohtse, 2023-09-30:
> Not sure if I'm missing something but I can't reproduce this on latest git build.

set chan 1 to AM band

now set chan to to any FM   (You can switch the chans it makes no dfference)

Make sure your TX chan is the Fm one

Now trans

Now try trans  during or jjst after an AM RX

it will fail  and it will either continue to fail(warning text states TX disabled)  or after a 10-20 seconds you can TX again providing you had not RX any further AM band

OneOfEleven, 2023-10-01:
Is Dual-RX (Dual Watch) enabled, and if it is, which is set to the main one (solid black little triangle marker) ?

lohtse, 2023-10-01:
> Is Dual-RX (Dual Watch) enabled, and if it is, which is set to the main one (solid black little triangle marker) ?

For me with or withot dual watch it happens. the main channel wil always be FM for trans(Black triangle) I never trans on AM band... I have a proper HT for that

OneOfEleven, 2023-10-01:
The radio does as you say if dual-watch is enable, when stops on the AM channel thats when the TX is disabled when you try to transmit.

If I disable dual-watch, the 'problem' vanishes because you will always transmit on the FM channel.

OneOfEleven, 2023-10-01:
I'll have to see how the original QS firmware operates.

lohtse, 2023-10-01:
> I'll have to see how the original QS firmware operates.

Actually stated it is the same (see above)

OneOfEleven, 2023-10-01:
I can't reproduce this when dual-watch is disable :(

lohtse, 2023-10-01:
> I can't reproduce this when dual-watch is disable :(

hmmmmmmmmmmmmmm odd but ok..

lohtse, 2023-10-01:
> > Is Dual-RX (Dual Watch) enabled, and if it is, which is set to the main one (solid black little triangle marker) ?
> 
> For me with or without dual watch it happens. the main channel wil always be FM for trans(Black triangle) I never trans on AM band... I have a proper HT for that

OneOfEleven, 2023-10-01:
Try it now, just uploaded

lohtse, 2023-10-01:
> Try it now, just uploaded

Given the AM issues on current FW's  for me I am now using older comps, So even though there is the issue (but not a major one) I don't use newer builds....

OneOfEleven, 2023-10-01:
okey doke

bricky149, 2023-10-01:
I don't use DW, might be why I couldn't reproduce. Will try after the compile errors are fixed.

OneOfEleven, 2023-10-01:
There is still a slight problem with TX'ing when dual-watch is enabled. I'll find the bug, just trying to chill today, so being lazy :)