OneOfEleven/uv-k5-firmware-custom #67: Set AGC routine.

INDIAJULIETVICTOR, 2023-09-28:
Hi, 
in doing some tests I noticed that the BK4819_SetAGC (1) routine makes the radio work much better, I saw that in this section there are calls to undocumented registers, do you have any information about these ?
Thanks Lot.

OneOfEleven, 2023-09-28:
No I wish I did. I don't know about any undocumented register info.

DualTachyon is the author, I suppose he'll know ?

INDIAJULIETVICTOR, 2023-09-28:
the best performance you have in transmission, modulation improves a lot, and reception seems more sensitive

INDIAJULIETVICTOR, 2023-09-28:
also in UI_UpdateRSSI, I made this correction which makes the RSSI graph the most linear, each notch now has an increase of 6 dB.

		const unsigned int band = gRxVfo->Band;
		const unsigned int offset = 10;
		const uint16_t level0  = gEEPROM_RSSI_CALIB[band][0]*2/4 + offset;
		const uint16_t level1  = gEEPROM_RSSI_CALIB[band][1]*3/5 + offset;
		const uint16_t level2  = gEEPROM_RSSI_CALIB[band][2]*5/7 + offset;
		const uint16_t level3  = gEEPROM_RSSI_CALIB[band][3]*7/9 + offset;

DE8MSH, 2023-09-28:
INDIAJULIETVICTOR : Again. Why dont you join https://t.me/quansheng_uvk5_en_dev on Telegram? Here are all the devs.

lohtse, 2023-09-28:
> INDIAJULIETVICTOR : Again. Why dont you join https://t.me/quansheng_uvk5_en_dev on Telegram? Here are all the devs.

He did but left and it is his choice to not return... please don't be pushy....

INDIAJULIETVICTOR, 2023-09-28:
I didn't want to offend anyone or even be intrusive, I just posted a suggestion.. 
Sorry for the inconvenience.

INDIAJULIETVICTOR, 2023-09-30:
no longer needed with new updates

closed 2023-09-30 by INDIAJULIETVICTOR