OneOfEleven/uv-k5-firmware-custom #68: AM issue...

lohtse, 2023-09-29:
This is OEFW-d449cc0

https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/48331155/f398382f-6d1d-4176-bc9c-eaac7406105a

lohtse, 2023-09-29:
this is OEFW-895d83e(latest build)

https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/48331155/157d52c0-f641-446b-8057-ade4c5261ee2

lohtse, 2023-09-29:
Please note the last 20seconds of OEFW-895d83e

also the constant rx led flashing

seems to nolonger pick up certain signal strengths   where as the first version(video) has no issues

lohtse, 2023-09-29:
The issue actually started around FW OEFW-c4dc895   and has steadily worsened...   I didn't report it as I thought it culd have been weather related but yesterday was good weather here and to day is similar so far.. so not a weather issue. But then as you can see from the two videos there is a clear differnce/issue

OneOfEleven, 2023-09-29:
It looks and sounds like you have some local RFI/EMI that comes and goes.

A worsening problem that most hams will tell you about from things like LCD screens, solar panel inverters, PC's, laptops, mains switching power supplies (inc USB chargers), LED lights, mains power line network adapters, you name it.

lohtse, 2023-09-29:
> It looks and sounds like you have some local RFI/EMI that comes and goes.
>
> A worsening problem that most hams will tell you about from things like LCD screens, solar panel inverters, PC's, laptops, [...]

issue with that is if I swutch back to firmware in first video the issue goes away..  So if it was the factors you suggest,  they would continue. yet they don't

OneOfEleven, 2023-09-29:
That's very true.

When it happens try walking with the radio say to opposite side of room or by window etc, my curiosity gets the better of me ;)

lohtse, 2023-09-29:
> That's very true.
> 
> When it happens try walking with the radio say to opposite side of room or by window etc, my curiosity gets the better of me ;)

no differnce..   same issues as above

lohtse, 2023-09-29:
and this is the same as on the very latest build as of this comment

lohtse, 2023-09-29:
https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/48331155/9b290fca-0cd5-4716-a5c1-d2285466e6e2

Latest build

lohtse, 2023-09-29:
https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/48331155/22f84210-deee-45b5-87b5-79aefd59abb7

build OEFW-d449cc0

lohtse, 2023-09-29:
videos are taken around 5mins apart

OneOfEleven, 2023-09-29:
I can't test it today as there are no useful airband comms that I can find today.

If I put my radios near my dell laptop they become so swamped with EMI/RFI they become useless.

lohtse, 2023-09-29:
no antema -118 s1

antenna -92 s5

OneOfEleven, 2023-09-29:
This is how it is for me - when near the laptop and it's screen. The 1st half is with antenna off, the 2nd half is with antenna on - the noise/RFI jumps from S1 to S7 for me. I need to know what yours does.

https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/51590168/e5e51855-cc5c-4b46-bb3f-aad1c9b9569d

OneOfEleven, 2023-09-29:
OK, thanks

lohtse, 2023-09-29:
> OK, thanks

do you need anything else?

lohtse, 2023-09-29:
OEFW-f4643ad(latest as of this message)

https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/48331155/99d3725e-7715-4f75-abd2-f68da69d7d0a

 notice the heavy static   as if it is trying to receive a signal

lohtse, 2023-09-29:
OEFW-d449cc0 FW  from two days ago(same as one earlier)

https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/48331155/2a163706-7ea8-4bcf-9923-7e7a66adcb40

Notice lack of sheavy static etc clear coms too

Both vids taken same location a few mins apart(time it takes to reflash)

But huge differnces....

lohtse, 2023-09-30:
Well as of *OEFW-aba9e43' fw   I can barely receive any AM indoors now  and similar outdoors unless I tracel around 3km away!!!!   The issue shown in the video two comments up  just got worse..https://github.com/OneOfEleven/uv-k5-firmware-custom/issues/68#issuecomment-1741472446

Flashed back to the above FW and lots of lovely AM traffic

If it was an  of the following  "LCD screens, solar panel inverters, PC's, laptops, mains switching power supplies (inc USB chargers), LED lights, mains power line network adapters"  or poor Signals etc it would continue across FW's

So I am now at a loss as to whats going on here.

OneOfEleven, 2023-10-01:
I have been looking, though been having a lazy few days.

I haven't actually changed the AM code for a while now, so it's a bit confusing as to why it's getting worse, baffling to me too. All I've really done something with past few days is two DTMF fixes that I broke and the S-meter.

lohtse, 2023-10-01:
> I have been looking, though been having a lazy few days.
> 
> I haven't actually changed the AM code for a while now, so it's a bit confusing as to why it's getting worse, baffling to me too. All I've really done something with past few days is two DTMF fixes that I broke and the S-meter.

Makes two of us lol... But even the vids above show some thinh has changes some where...just no idea what

lohtse, 2023-10-01:
PS

keep meaning to ask what does thi actually do?

ENABLE_VOICE                  := 0       want to hear voices ?

OneOfEleven, 2023-10-01:
Setting it to '0' disables the voices (the incomplete spoken chi/eng stuff), saves a good bit of flash space.

lohtse, 2023-10-01:
> Setting it to '0' disables the voices (the incomplete spoken chi/eng stuff), saves a good bit of flash space.

lol did even notice it was off when used..

anyway could you compare the twp files in the Zip... one is the last good am  FW and the other is the latest Comp.
[OneOfEleven K5 lCustom FW.zip](https://github.com/OneOfEleven/uv-k5-firmware-custom/files/12776791/OneOfEleven.K5.lCustom.FW.zip)

OneOfEleven, 2023-10-01:
Apart from the old firmware having bugs in the gain setting they sound very much the same.

Old firmware on the left (white), new on the right (yellow). You won't hear it in the vid but I hear they are the same, though the new one has the BK4819 auto RX bandwidth weak signal switching enabled (I was asked to reenable it).

Where I live the signals aren't very strong though, so testing stronger signal handling takes me days while waiting for rare over-head fly overs.

https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/51590168/a39282f6-5ed5-42e3-a915-264bf210a2dd

lohtse, 2023-10-01:
Hmmm the plot thickens then..I get great AM everywhere with earlier FW but since then as seen in vids its poor to none on builds..

lohtse, 2023-10-01:
Curiously just flashed bricky,s fork of yours fw and it works  like older one i sent.. his fork is 2days old etc.

lohtse, 2023-10-01:
https://github.com/bricky149/uv-k5-firmware-custom

bricky149, 2023-10-01:
> Curiously just flashed bricky,s fork of yours fw and it works like older one i sent.. his fork is 2days old etc.

Gives me an idea of where to look. If I find it I'll do a PR, no doubt 1o11 will beat me as always. :)

OneOfEleven, 2023-10-01:
One thing I've noticed is that the K5(8) and/or the K6 speaker audio quality is better than the original K5. but I suspect thats due to a speaker change and/or plastic casing change.

bricky149, 2023-10-01:
I think I've just reproduced it.

When I turn the radio on, my local ATIS is received with a gain of -7. After the antenna touches my laptop lid (metal) and put the radio back where I had it when I turned it on, the gain becomes -4 and the AM signal is improved. Rebooting the radio without moving it returns the gain back to -7, and the signal returns to what it was.

Could that be it?

lohtse, 2023-10-01:
> I think I've just reproduced it.
> 
> When I turn the radio on, my local ATIS is received with a gain of -7. After the antenna touches my laptop lid (metal) and put the radio back where I had it when I turned it on, the gain becomes -4 and the AM signal has more static. Rebooting the radio without moving it returns the gain back to -7, and the signal returns to what it was.
> 
> Could that be it?

I would love to think that was the case... But why would that only effect certain FWs ??

OneOfEleven, 2023-10-01:
> I think I've just reproduced it.
>
> When I turn the radio on, my local ATIS is received with a gain of -7. After the antenna touches my laptop lid (metal) and put the radio back where I had it when I [...]

That's just a bit of hysteresis in the AM AGC loop I use to help prevent gain hunting. 3 or 4dB is fine, it wouldn't cause distortion.

If I could reproduce the problem, I could find it, but I just can't seem to get any firmware version to break.

lohtse, 2023-10-01:
The CHanges you just made seem to have solved  the issue.. There's abit more static than from the earlier one in zip above but its pretty much on par with that version now.

Thank you

troilus, 2023-10-03:
A user reported that after updating 1O11 firmware, the AIR frequency had strong back noise when receiving and could hardly hear the voice, and finally found that although the VFO showed AM, it was actually still FM mode , and finally he went to the menu to re-set to AM then work correctly.

I asked him about some of the steps at the time, and he only said that the AIR frequency was stored in the channel, but he couldn't recall the steps, and we couldn't reproduce the problem.

troilus, 2023-10-03:
The radio on the right is the state at the time of the problem. It should be a small probability problem under complex operation steps, So I asked him to keep trying to reproduce the problem 100% to give some help.

Uploading 1.mov…

OneOfEleven, 2023-10-03:
Ask them if they have the spoken audio voices enabled or not ?

OneOfEleven, 2023-10-03:
If it's been left in FM mode on occasion then that could explain a few things.

troilus, 2023-10-03:
> Ask them if they have the spoken audio voices enabled or not ?
> 
> 

Voice is disabled while compiling.

OneOfEleven, 2023-10-03:
oh, darn, thought I'd found the problem.

OneOfEleven, 2023-10-03:
> > Ask them if they have the spoken audio voices enabled or not ?
> 
> Voice is disabled while compiling.

Some ones telling fibs, I can't fix things if people give me disinformation.

The voice icon is showing on the one radio, which can't happen if voices option is not compiled in. But anyway, I'll still look for a possible problem, but I need to reproduce it.

troilus, 2023-10-04:
Im sorry and I tried  to reproduce it, have tried save FREQ with CPS and AM flag enable or disable but still unable to reproduce it so far.

closed 2023-10-20 by OneOfEleven