OneOfEleven/uv-k5-firmware-custom #71: Rssi sbar not work

Braccato, 2023-09-29:
since I installed the latest firmware with the rssi bar the display has been damaged

https://github.com/OneOfEleven/uv-k5-firmware-custom/assets/141162772/15399f5e-4738-40f9-86d6-167ca48955e9

lohtse, 2023-09-29:
I suggest you go to telegram group as this is a KNOWN issue.

Braccato, 2023-09-29:
link telegram please

OneOfEleven, 2023-09-29:
That doesn't look good !

Looks like a hardware problem there, maybe dropped the radio or something ?

lohtse, 2023-09-29:
https://t.me/quansheng_uvk5_en

Braccato, 2023-09-29:
the radio is new, I also tested it on the k5 and the 5R_plus, it has the same defect

lohtse, 2023-09-29:
> the radio is new, I also tested it on the k5 and the 5R_plus, it has the same defect

NOT a DEFECT a bug.. And now I remember who you are...

lohtse, 2023-09-29:
there was another member on telegram who did the same to produce the bug... literally the only way it happens is when they do the same covering up and pressing buttons at the same time.

lohtse, 2023-09-29:
I would close this as it is not anb  issue with your FW.

OneOfEleven, 2023-09-29:
That's a design problem with the radio, RF getting into where you don't want it too.

Braccato, 2023-09-29:
it is a flaw of all rssi sbar, you have to find a solution, even with pietr022's rssi sbar it does it

OneOfEleven, 2023-09-29:
Nothing I can do, it's an RF problem.

lohtse, 2023-09-29:
> it is a flaw of all rssi sbar, you have to find a solution, even with pietr022's rssi sbar it does it

Yes TORRE JOHN... hell why don't YOU FIX IT

OneOfEleven, 2023-09-29:
Some decoupling caps places in some suitable positions on the PCB might help, but I'll leave that up to everyone else.

closed 2023-09-29 by OneOfEleven