OneOfEleven/uv-k5-firmware-custom #72: dtmf decoder losts first character of dtmf string

pawol, 2023-09-29:
dtmf decoder losts first character of dtmf string.

Reproduction:
d-live on
press * 0123 PTT on tx side
on rx side we get 123 only. 
Problem occurs inly when rx side receives dtmf first time. Next transmissions are ok. I mean that probably character on 1st position in line on screen on rx side is lost

OneOfEleven, 2023-09-29:
I must have broke it. Try new upload now.

pawol, 2023-09-29:
Similarily . Sent 012345 but received 12345 when 1st time used. When a few times dtmf is send then string is properly received rather every time

pawol, 2023-09-29:
I noticed that it can be Tx problem.I was trying to send 2222 and I received only 2. I heard only one tone !

OneOfEleven, 2023-09-30:
I can see whats happening now, bare with me.

Just been locating a couple of major bugs in the DTMF code. It's fun to use once you know how to use it for calling/paging other radios.

pawol, 2023-09-30:
On the latest  build 9493f34 the primary issue seems to be fixed

OneOfEleven, 2023-09-30:
The missing initial decoding problem is still there, I haven't do that yet, it's fixed after the first squelch opening/closing or first transmit-receive switch you do, It must just be me not properly resetting/setting the DTMF setting in the chip.

I'll get around to it in a day or two, just don't want to at the mo, am chilling :)

pawol, 2023-09-30:
Great you found root cause

pawol, 2023-10-02:
The issue  seems to be fixed on commit e0aa1a8

pawol, 2023-10-04:
The issue still occurs on 429ac36

OneOfEleven, 2023-10-04:
I haven't tackled it, keep getting diverted to other subroutines. I will arrive back to fix it though.

lohtse, 2023-10-04:
> The issue still occurs on 429ac36

Jeez give poor Dev chance to rest...don't forget they  do this for free and in their free time.. They do have lives too...

pawol, 2023-10-04:
Got it Fast scanning impacts on squelch (PR #117) and on this DTMF case
On normal scanning option everything works okay - squelch opened by 1750 and missing 1st character in DTMF string works

pawol, 2023-10-05:
The issue fixed on e042b21. PR closed

closed 2023-10-05 by pawol