OneOfEleven/uv-k5-firmware-custom #74: About the accuracy of Sbar

troilus, 2023-09-30:
Thanks for your mod firmware!

According to this part of the code in ui/main.c, and the algorithm mentioned in [https://www.giangrandi.org/electronics/radio/smeter.shtml](https://www.giangrandi.org/electronics/radio/smeter.shtml), perhaps the current S_level calculation method only applies to frequencies below 30MHz.

It's just some information that I personally learned, not necessarily correct, if there is a mistake, please to correct
```
#if defined(ENABLE_RSSI_BAR)
	void UI_DisplayRSSIBar(const int16_t rssi, const bool now)
	{
		char               s[16];
		unsigned int       i;

		const unsigned int max_dB      = -33;
		const unsigned int min_dB      = -127;

		const unsigned int txt_width   = 7 * 10;         // 10 chars
		const unsigned int bar_x       = txt_width + 4;
		const unsigned int bar_width   = LCD_WIDTH - 1 - bar_x;

		const int16_t      dBm         = (rssi / 2) - 160;
		const unsigned int clamped_dBm = (dBm < min_dB) ? min_dB : (dBm > max_dB) ? max_dB : dBm;
		const unsigned int width       = max_dB - min_dB;
		const unsigned int len         = (((clamped_dBm - min_dB) * bar_width) + (width / 2)) / width;

		const unsigned int line        = 3;
		uint8_t           *pLine       = gFrameBuffer[line];

		unsigned int       s_level     = 0;

		if (gEeprom.KEY_LOCK && gKeypadLocked > 0)
			return;     // display is in use
		if (gCurrentFunction == FUNCTION_TRANSMIT || gScreenToDisplay != DISPLAY_MAIN)
			return;     // display is in use

		if (dBm >= -63)                    // S9+10dB
			s_level = 10 + (((dBm - -63) / 10) * 10);
		else
		if (clamped_dBm >= -127)           // S0
			s_level = (clamped_dBm - -127) / 6;

		if (now)
			memset(pLine, 0, LCD_WIDTH);

		if (s_level < 10)
			sprintf(s, "%-4d S%u  ", dBm, s_level);  // S0 ~ S9
		else
			sprintf(s, "%-4d S9+%u", dBm, s_level);  // S9+
```

OneOfEleven, 2023-09-30:
That code you're using there is a bit behind where I am at the moment. I made it easier to change the S-levels yesterday.

troilus, 2023-09-30:
Yes, but I mean the algorithm of the s_level differs depending on whether the frequency is greater than 30MHz according to the url.

OneOfEleven, 2023-09-30:
The algorithms the same, still 6dB/S-point up to S9, there after it's +10dB, +20dB etc (or whatever you prefer).

I made the the S0 and S9 dB levels easily changeable, even with frequency if you like (need a bit of extra code).
Anyway, I don't use S-points, I much prefer dBm and/or uV, they are absolute.

zhaoparrot, 2023-09-30:
@troilus do u think this
```
		const int16_t      s9_dBm       = -93;                   // S9
		const int16_t      s0_dBm       = -147;                  // S0
```
would be more accurate?

OneOfEleven, 2023-09-30:
It's pointless setting the S-meter anywhere below about -120dBm anyway, if this graph of RX sensitivity is anywhere near accurate then the S-reading would never go below S4'ish if you set S0 to -147dBm ..

[https://mikhail-yudin.ru/notes/quansheng-uv-k5-opyt-raboty/IMG_20230705_232055_209.jpg](url)

OneOfEleven, 2023-09-30:
The minimum discernible signal on 145MHz I've with my radio(s) is about -120dBm (S1'ish).

12dB sinad I can't measure as I don't have a noise meter here, but at a guess the radios quieting level sounds about right at around -118dBm (matches well).

My radios just don't detect anything below the minimum discernible value I mention above.

If others can do the same test it will help.

closed 2023-10-03 by OneOfEleven