OneOfEleven/uv-k5-firmware-custom #75: [Suggestion] Use horizontal menu to display menu names more clearly.

troilus, 2023-09-30:
😋  example   https://github.com/rebezhir/openquack/blob/main/ui/menu.c

OneOfEleven, 2023-09-30:
Yes I've been meaning to make the menu better from the word go. Just been side tracked by lots of other stuff.

OneOfEleven, 2023-10-03:
The menu seems to be slowly morphing, not sure what into though.

ea7hjb, 2023-10-05:
Since this "issue" is related with the menu, trying not to open a new ticket, I will coment here that I observed that activating the keypad lock (Long press F key) is possible while inside the menu. Is that normal?

OneOfEleven, 2023-10-05:
I'll fix that, it's a side effect of me not closing the menu whilst the user is transmitting (useful to change a menu setting between TX's).

OneOfEleven, 2023-10-05:
I thought about it a while back but decided not too as I'm also not keen on that particular style. I'm still playing with the menu style to see if I can make it easier to use / cleanest looking as best I can.

closed 2023-10-05 by OneOfEleven