OneOfEleven/uv-k5-firmware-custom #76: When transmitting or receiving the 409 frequency, only the LED light is on and there is no sound

CaptainRi, 2023-09-30:
When transmitting or receiving the 409 frequency, only the LED light is on and there is no sound

pawol, 2023-09-30:
Do you mean transmission on 409MHz?

pawol, 2023-09-30:
I confirm bad behaviour on 409MHz on the latest build 9493f34.
One on my  QS transmits something but second QS receives very distorted audio.

In opposite direction QS tells me "TX disallowed" but both of QSs has the same config.
I saw spectrum and 409MHz is clear in my location

pawol, 2023-09-30:
Update. Everything works okay. I had AM set in on in my QS and bandwidth was also different. Forget abou my previous report

CaptainRi, 2023-10-01:
The reason has been found, I used non-standard sub tones

OneOfEleven, 2023-10-01:
ah ha, that's why I couldn't make the radio not work at that frequency.

I find pictures/video (of the radios LCD screen) tend to let me see what's really going on, words alone tend not to give the whole picture and so make it impossible to diagnose the problem.

closed 2023-10-01 by OneOfEleven