OneOfEleven/uv-k5-firmware-custom #77: Open RSSI threshold lower than close when reduced Squelch1 values are used above airband

bricky149, 2023-09-30:
Regarding this:
https://github.com/OneOfEleven/uv-k5-firmware-custom/issues/65#issuecomment-1739796727

Firstly, thanks for responding to my query by proxy. I think there's a misunderstanding as to what I intended to say.

From radio.c:468 where the selected frequency is in airband:
					pInfo->SquelchOpenRSSIThresh   = ((uint16_t)pInfo->SquelchOpenRSSIThresh     * 11) / 12;
					pInfo->SquelchCloseRSSIThresh  = ((uint16_t)pInfo->SquelchOpenRSSIThresh     *  9) / 10;

The above has the open threshold higher than the close threshold. (11/12 = 0.916, 9/10 = 0.9)

Same file, from where the band is higher than airband:
					pInfo->SquelchOpenRSSIThresh   = ((uint16_t)pInfo->SquelchOpenRSSIThresh     *  3) /  4;
					pInfo->SquelchCloseRSSIThresh  = ((uint16_t)pInfo->SquelchOpenRSSIThresh     *  9) / 10;

The above has the open threshold lower than the close threshold. (3/4 = 0.75, 9/10 = 0.9)

The commented values suggest only the first example is right. Am I understanding this correctly?

OneOfEleven, 2023-09-30:
oh right, I see.

I'll have to fix that.

OneOfEleven, 2023-09-30:
Think I got it right now :)

I can lower the squelch-1 threshold more, but difficult to know just how low to go without peoples squelches becoming too sensitive/annoying for them.

OneOfEleven, 2023-09-30:
Thank you by the way.

closed 2023-09-30 by OneOfEleven