OneOfEleven/uv-k5-firmware-custom #78: hat's the meaning of 96.96?

CaptainRi, 2023-10-01:
[image]
what's the meaning of 96.96?hat's the meaning of 96.96?

OneOfEleven, 2023-10-01:
It's debug data that I'm using, it actually means index 96 from a total of 96 entries. The AM look-up table has 96 entries (2nd figure) and the code is currently using the last entry (1st figure) number 96.

The higher the 1st figure the higher gain the front end is set too..'1' being to lowest gain, '96' being the highest gain.

You can disable the showing of that AM test data by changing this line in Makefile from this ..

ENABLE_AM_FIX_SHOW_DATA       := 1

to this ...

ENABLE_AM_FIX_SHOW_DATA       := 0

then recompile.

CaptainRi, 2023-10-01:
Thank you very much for your answer. You are so helpful, thank you very much

closed 2023-10-01 by CaptainRi