OneOfEleven/uv-k5-firmware-custom #79: Cannot compile the latest build b45043a

pawol, 2023-10-01:
Cannot compile the latest build b45043a. Failed as below

-I c:/uv-k5-firmware-custom/ -I c:/uv-k5-firmware-custom//external/CMSIS_5/CMSIS/Core/Include/ -I c:/uv-k5-firmware-custom//external/CMSIS_5/Device/ARM/ARMCM0/Include -c ui/lock.c -o ui/lock.o
arm-none-eabi-gcc -Os -Wall -Werror -mcpu=cortex-m0 -fno-builtin -fshort-enums -fno-delete-null-pointer-checks -std=c11 -MMD -DPRINTF_INCLUDE_CONFIG_H -DGIT_HASH=\"b45043a\" -DENABLE_FMRADIO -DENABLE_OVERLAY -DENABLE_UART -DENABLE_SMALL_BOLD -DKEEP_MEM_NAME -DENABLE_WIDE_RX -DENABLE_CTCSS_TAIL_PHASE_SHIFT -DENABLE_MAIN_KEY_HOLD -DENABLE_COMPANDER -DENABLE_REVERSE_BAT_SYMBOL -DENABLE_AM_FIX -DENABLE_AM_FIX_SHOW_DATA -DENABLE_SQUELCH_LOWER -DENABLE_RSSI_BAR -DENABLE_AUDIO_BAR  -I c:/uv-k5-firmware-custom/ -I c:/uv-k5-firmware-custom//external/CMSIS_5/CMSIS/Core/Include/ -I c:/uv-k5-firmware-custom//external/CMSIS_5/Device/ARM/ARMCM0/Include -c ui/main.c -o ui/main.o
ui/main.c: In function 'UI_DisplayMain':
ui/main.c:422:39: error: 'Line' undeclared (first use in this function)
  422 |      UI_PrintStringSmall("TX", 14, 0, Line);
      |                                       ^~~~
ui/main.c:422:39: note: each undeclared identifier is reported only once for each function it appears in
ui/main.c:500:9: error: 'Channel' undeclared (first use in this function); did you mean 'channel'?
  500 |     if (Channel == vfo_num)
      |         ^~~~~~~
      |         channel
ui/main.c:321:22: error: unused variable 'rx_line' [-Werror=unused-variable]
  321 |   const unsigned int rx_line  = (gEeprom.RX_CHANNEL == 0) ? line0 : line1;
      |                      ^~~~~~~
cc1.exe: all warnings being treated as errors
make: *** [ui/main.o] Błąd 1
Press any key to continue . . .

bricky149, 2023-10-01:
The dev is aware of this issue:
https://github.com/OneOfEleven/uv-k5-firmware-custom/commit/b45043aab292ca46583c66e24533f17abcd0637a#commitcomment-128833887

OneOfEleven, 2023-10-01:
Hopefully fix

closed 2023-10-01 by OneOfEleven