OneOfEleven/uv-k5-firmware-custom #82: On commit c3c344d sometime tx is allowed when AM

pawol, 2023-10-01:
On commit c3c344d sometime tx is allowed when AM.
Just after update.
Pressing of PTT does not show TX disable and red led is lit. After a few attempt feature start to work

OneOfEleven, 2023-10-01:
Yes I know, I've been trying to do too many things at once,, I'll fix it shortly. Bare with me.

pawol, 2023-10-01:
Just report you ti keep it in mind :)

OneOfEleven, 2023-10-01:
ha. software/firmware is NEVER finished.

Just don't delete this issue, otherwise I will forget :)

OneOfEleven, 2023-10-01:
Do you have dual-watch (Dual-RX) enabled when the problem happens ?

lohtse, 2023-10-01:
> On commit c3c344d sometime tx is allowed when AM.
> Just after update or QS power off/on.
> Pressing of PTT does not show TX disable and red led is lit. It seems the behaviour occurs until 1st squelch opening.

I had reported this already buddy..you could have just added too that instead of starting  new issue...

OneOfEleven, 2023-10-01:
>**It seems the behaviour occurs until 1st squelch opening.**

Thank you !   that's a key bit of info, I'll soon find that.

pawol, 2023-10-01:
Yes, I have dual watch and active channel is set on AM

pawol, 2023-10-01:
Since 2 commits I use lower squelch feature also

OneOfEleven, 2023-10-01:
it's only when dual-watch is enabled. It does transmit on the FM channel not the AM channel, it's just that it shouldn't be transmitting at all when the main channel is the AM one.

pawol, 2023-10-01:
I shall check it with standard squelch. I never noticed it earlier

pawol, 2023-10-01:
I have checked and the issue depends only on dual watch

OneOfEleven, 2023-10-01:
Try it now, I 'think' I've got it, though there are multiple factors that affect the TX freq/chan to take into account, my heads spinning from it.

OneOfEleven, 2023-10-01:
no, ignore that :(

pawol, 2023-10-02:
The issue is fixed on commit e0aa1a8. 
The ticket can be closed

OneOfEleven, 2023-10-02:
i think there is still a funny with, I had it do it once which means I've missed something that needs finding.

pawol, 2023-10-04:
IMHO the bug was fixed. Cannot reproduce the issue.

OneOfEleven, 2023-10-05:
I've been changing it slightly, so let me know if the problem re-appears, or if it TX's on the wrong VFO (should only ever TX on the selected VFO - solid black triangle indicator).

pawol, 2023-10-05:
Cannot reproduce now

closed 2023-10-05 by pawol